embassy-rs/embassy #2418: Can't compile blinky example on stm32h503rb chip

NathanYee, 2024-01-08:
I am new-ish to embedded Rust so it is entirely possible I'm misunderstanding how to change a chip in the examples. Here is what I did.

After changing `embassy/examples/stm32h5/Cargo.toml` to

```toml
# embassy-stm32 = { version = "0.1.0", path = "../../embassy-stm32", features = ["defmt", "stm32h563zi", "memory-x", "time-driver-any", "exti", "unstable-pac"] }
embassy-stm32 = { version = "0.1.0", path = "../../embassy-stm32", features = ["defmt", "stm32h503rb", "memory-x", "time-driver-any", "exti", "unstable-pac"] }
```

and `embassy/examples/stm32h5/.cargo/config.toml` to
```toml
# runner = 'probe-rs run --chip STM32H563ZITx'
runner = 'probe-rs run --chip STM32H503RBTx'
```

Running `cargo build --release --bin blinky` gives errors:
```
error: linking with `rust-lld` failed: exit status: 1
...
  = note: rust-lld: error: section '.bss' will not fit in region 'RAM': overflowed by 564 bytes
          rust-lld: error: section '.uninit' will not fit in region 'RAM': overflowed by 1588 bytes
          

error: could not compile `embassy-stm32h5-examples` (bin "blinky") due to previous error
```

Dirbaio, 2024-10-18:
Try lowering the task arena size of the executor. https://docs.embassy.dev/embassy-executor/git/std/index.html#task-arena

Or if you're OK with nightly, enable feature `nightly` which makes the tasks occupy exactly the space they need, nothing more.

ArXen42, 2024-10-18:
> Try lowering the task arena size of the executor. https://docs.embassy.dev/embassy-executor/git/std/index.html#task-arena
> 
> Or if you're OK with nightly, enable feature `nightly` which makes the tasks occupy exactly the space they need, nothing more.

Thanks! It compiles now with custom arena size, for some reason I didn't think to check how tasks are allocated.

Can't get nightly to compile for me yet (missing waker.vtable method), but I will try some more later.