embassy-rs/embassy #2425: embassy-net 3.0 doesn't compile on esp32 due to async_fn_in_trait  and impl_trait_projections

yanshay, 2024-01-10:
I'm trying to upgrade a working esp32s3 project to embassy-net 3.0 (forced due to some dependency requirements) and face the errors below.
I added the suggested features to my crate, but it doesn't make any difference.
Seems like the rust compiler doesn't support these attributes (though it doesn't throw any error).

I tried it also on a new no_std project, just added 
`embassy-net = { version = "0.3.0", features = ["tcp", "udp", "dhcpv4", "medium-ethernet"]}` and those errors appeared.

Am I doing something wrong or isn't esp32 working any longer with new embassy-net?

```
error[E0706]: functions in traits cannot be declared `async`
   --> /Users/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-net-0.3.0/src/tcp.rs:545:9
    |
545 |         async fn read(&mut self, buf: &mut [u8]) -> Result<usize, Self::Error> {
    |         -----^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |         |
    |         `async` because of this
    |
    = note: `async` trait functions are not currently supported
    = note: consider using the `async-trait` crate: https://crates.io/crates/async-trait
    = note: see issue #91611 <https://github.com/rust-lang/rust/issues/91611> for more information
    = help: add `#![feature(async_fn_in_trait)]` to the crate attributes to enable
```

and 
```
error[E0658]: `async fn` return type cannot contain a projection or `Self` that references lifetimes from a parent scope
   --> /Users/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-net-0.3.0/src/tcp.rs:557:50
    |
557 |         async fn write(&mut self, buf: &[u8]) -> Result<usize, Self::Error> {
    |                                                  ^^^^^^^^^^^^^^----^^^^^^^^
    |                                                                |
    |                                                                help: consider spelling out the type instead: `embedded_io_impls::<impl embedded_io_async::Write for tcp::TcpSocket<'d>>`
    |
    = note: see issue #103532 <https://github.com/rust-lang/rust/issues/103532> for more information
    = help: add `#![feature(impl_trait_projections)]` to the crate attributes to enable
```

lulf, 2024-01-10:
You need to update your compiler to latest stable (1.75) or newer.

yanshay, 2024-01-10:
I think with esp I have to use their version of the compiler, at least as far as I know - https://github.com/esp-rs/rust.
Is there a way to upgrade something there? Isn't it a single package of the compiler?

Also, isn't upgrading compiler just instead of adding the attributes manually? (unrelated to the esp problem)

Dirbaio, 2024-01-10:
check out esp's documentation I believe running `espup update` should be enough to update.

yanshay, 2024-01-10:
> check out esp's documentation I believe running `espup update` should be enough to update.

Thanks, it worked.

closed 2024-01-10 by Dirbaio