embassy-rs/embassy #2431: No TIM2 on STM32L052

shufps, 2024-01-11:
Hi,

I tried to setup PWM for TIM2, Channel 1 and the compiler complains that it doesn't know TIM2.

Error:

```
error[E0609]: no field `TIM2` on type `Peripherals`
   --> src/bin/blinky.rs:155:36
    |
155 |     let mut pwm = SimplePwm::new(p.TIM2, Some(ch1), None, None, None, khz(10), Default::default());
    |                                    ^^^^ help: a field with a similar name exists: `TIM21`

```

and it doesn't seem to exist?

[image]

But TIM2 appears several times in the `_generated.rs`.

Has someone an idea what I'm doing wrong? :pleading_face:

shufps, 2024-01-11:
Hmm, TIM2 appears in `peripherals_definition` but not in `peripherals_struct` :thinking:

shufps, 2024-01-11:
and it also seems to appear in the included metadata.rs:

```
    Peripheral {
        name: "TIM2",
        address: 1073741824,
        registers: Some(PeripheralRegisters {
            kind: "timer",
            version: "v1",
            block: "TIM_GP16",
            ir: &timer::REGISTERS,
        }),
```

timokroeger, 2024-01-11:
Do you use the `time-driver-any` feature of `embassy-stm32`? If I remember correctly it picks TIM2 as backing timer making it unavailable to use in the application code.
You can either disable the time feature altogether or make it use a different timer with [one of those cargo features](https://docs.embassy.dev/embassy-stm32/git/stm32h743zi/index.html#time)

shufps, 2024-01-11:
Yes, `time-driver-any` is the default in all examples.

Okay, I see! I'll switch it to hmm ...`tim6` then because it's unused.

Thx a lot!

shufps, 2024-01-11:
Oh, there is no `time-driver-tim6` feature :sob: 

This will require a hardware change because I need it :pleading_face:

timokroeger, 2024-01-11:
> Oh, there is no time-driver-tim6 feature
It looks like TIM6 has no compare channel which is required to use it as timer driver

> Why doesn't it use the systick timer like on STM32 HAL?
https://github.com/embassy-rs/embassy/issues/786

shufps, 2024-01-11:
@timokroeger it seems that `tim22` is working fine - and probably `tim21` will work too.

Will do a PR for extending the feature-set and build.rs script :blush: