embassy-rs/embassy #2458: embassy-rp: uart driver looses errors when reading a simple byte

enbyted, 2024-01-17:
Given the following example:
```rust
    let mut buf = [0; 1];
    debug!("Waiting for bytes");
    loop {
        // system_bus is Uart<'static, UART0, uart::Async>
        if let Err(e) = system_bus.read(&mut buf).await {
            error!("uart read error {e:?}");
        } else {
            debug!("Received byte 0x{:02X}", buf[0]);
        }
    }
```

And that UART0 is connected to a source that transmits `<BREAK> 1 2 3` I would expect the following to be printed:
 - error Break
 - byte 1
 - byte 2
 - byte 3

However actually printed is:
 - byte 0
 - byte 1
 - byte 2
 - byte 3

After a bit of debugging it appears that there is a race condition in the resolution of this `select`:
```rust
/// embassy-rp/src/uart/mod.rs:415
        // wait for either the transfer to complete or an error to happen.
        let transfer_result = select(
            transfer,
            poll_fn(|cx| {
                T::dma_state().rx_err_waker.register(cx.waker());
                match T::dma_state().rx_errs.swap(0, Ordering::Relaxed) {
                    0 => Poll::Pending,
                    e => Poll::Ready(Uartris(e as u32)),
                }
            }),
        )
        .await;
```

Or, more concretely the comment in the interrupt handler is not correct, at least not in case of single-byte requests.
```rust
        let uart = T::regs();
        if !uart.uartdmacr().read().rxdmae() {
            return;
        }

        let state = T::dma_state();
        let errs = uart.uartris().read();
        state.rx_errs.store(errs.0 as u16, Ordering::Relaxed);
        state.rx_err_waker.wake();
        // disable the error interrupts instead of clearing the flags. clearing the
        // flags would allow the dma transfer to continue, potentially signaling
        // completion before we can check for errors that happened *during* the transfer.
        uart.uartimsc().write_clear(|w| w.0 = errs.0);
```

Debugging reveals that both interrupts - the UART and DMA - happen at almost the same time. The UART one is slightly faster (checked by setting them to the same dynamic priority, DMA has higher static priority <lower number>, yet UART is called first) likely meaning that there is a timing-issue in hardware with masking the DMA request line when error happens. 

At any rate, swapping the items in select seems to resolve the issue. I'm not sure if that is really "the fix", but in my very limited testing it works.

Maybe a more proper fix would be to disable the DMA channel and clear it's interrupt in the UART interrupt handler. Though that would infringe on DMA driver's responsibility.

Any thoughts? 
I can make a PR with a fix for this, after we come to an agreement on what is the best way to approach this issue :)

jamesmunns, 2024-01-19:
This *might* have been fixed by #2311. I caught this as well, and included the fix, though I might not have applied it everywhere.

Check this part out: https://github.com/embassy-rs/embassy/pull/2311/files#diff-ac33fe63c3c6d8b35edbd2ad57143020c9df7be1e5b6be8d7ca7eebd0ccbfd32R442-R453

jamesmunns, 2024-01-19:
@enbyted could you please try the current `main` branch, which has these fixes, and let me know if the issue still persists?

enbyted, 2024-01-19:
Hi @jamesmunns,

Yes, your change also fixes this issue.

jamesmunns, 2024-01-19:
@enbyted can this issue be closed then?

enbyted, 2024-05-03:
HI @jamesmunns,

Apparently this is not fully fixed, I still see this issue, however it is not 100% reproducible.

I don't have a minimal reproduction scenario yet, working on it.

## Usecase details

For now I'll share the details regarding my usage, maybe that will give you some ideas.

I have the following read pattern:
 1. Loop reading 1 byte at a time, waiting for break (dropping the bytes - a synchronization loop)
 2. Read of 1 byte
 3. Loop reading 1 byte at a time, using the bytes. Ends when break is received.
 4. Read of 1 byte
 5. Read of 1 byte
 6. Read of 12 bytes
 7. 100ms delay (on the sending side)

Sending side sends:
 - break
 - data for steps 1-3
 - break
 - data for steps 4-6
 - delay 100ms

Under these conditions exactly every other loop (i.e. 50% of the time), step no. 1 returns `0` instead of break error, while step no. 3 always receives the break correctly.

If I make loop in step no. 1 also break out when receiving a `0` then step no. 1 never sees a break error.

## Rootcause hypothesis

My gut feeling says that something weird happens in hardware when the DMA is set to 1 byte and the interrupt for uart errors is also set. Likely the correct solution would be to have a 16-bit DMA buffer that will also capture the error flags. In a past implementation of this protocol (also on rp2040, in C++) I actually had some parts of the decoding state machine directly in the uart interrupt, only setting the DMA when I knew there is a bulk transfer.

I suppose it is possible that the dmaonerr flag doesn't work reliably (especially given the original issue with the order of interrupts and the fact the the DMA transfer takes place at all). I don't see any relevant errata for this case, but further testing may be required. 

## Possible workaround for my particular usecase

In general, setting the DMA to transfer one byte a time is quite wasteful and likely the cause of the problems here. Maybe embassy could offer a one-shot uart-interrupt-driven (if the FIFO is empty on entry) receive special case when reading one byte at a time? Separate function for that usecase would be okay as well.

enbyted, 2024-05-04:
Some more breadcrumbs to follow.

I have added a log line in the read function:
```rust
        T::regs().uartdmacr().write_set(|reg| {
            reg.set_rxdmae(true);
            reg.set_dmaonerr(true);
        });

        trace!("Read starting, rx_errs: {:04X}, status reg: {:04X}, mask: {:04X}", T::dma_state().rx_errs.load(Ordering::Relaxed), T::regs().uartris().read().0, T::regs().uartimsc().read().0);

        let transfer = unsafe {
            // If we don't assign future to a variable, the data register pointer
            // is held across an await and makes the future non-Send.
            crate::dma::read(ch, T::regs().uartdr().as_ptr() as *const _, buffer, T::RX_DREQ)
        };

        // wait for either the transfer to complete or an error to happen.
        let transfer_result = select(
            transfer,
            poll_fn(|cx| {
                T::dma_state().rx_err_waker.register(cx.waker());
                match T::dma_state().rx_errs.swap(0, Ordering::Relaxed) {
                    0 => Poll::Pending,
                    e => Poll::Ready(Uartris(e as u32)),
                }
            }),
        )
        .await;

        let errors = match transfer_result {
            Either::First(()) => {
                // We're here because the DMA finished, BUT if an error occurred on the LAST
                // byte, then we may still need to grab the error state!
                Uartris(T::dma_state().rx_errs.swap(0, Ordering::Relaxed) as u32)
            }
            Either::Second(e) => {
                // We're here because we errored, which means this is the error that
                // was problematic.
                e
            }
        };

        trace!("Read finished, errors: {:04X}, rx_errs: {:04X}, status reg: {:04X}, mask: {:04X}", errors.0, T::dma_state().rx_errs.load(Ordering::Relaxed), T::regs().uartris().read().0, T::regs().uartimsc().read().0);

        // If we got no error, just return at this point
        if errors.0 == 0 {
            return Ok(());
        }
```

And in the error case it prints:
```
000006.015432 [T embassy_rp::uart] Read starting, rx_errs: 0000, status reg: 0020, mask: 0780
000006.316468 [T embassy_rp::uart] Read finished, errors: 0000, rx_errs: 0000, status reg: 02A0, mask: 0500
```

I have also added logs to the intrrupt handler itself and that log is not visible - supposedly it is not firing. I see that log in the correct case, example:
```
000005.076146 [T embassy_rp::uart] Read starting, rx_errs: 0000, status reg: 0020, mask: 0780
000005.376448 [I embassy_rp::uart] Uart interrupt, errs: 02A0
000005.376564 [T embassy_rp::uart] Read finished, errors: 02A0, rx_errs: 0000, status reg: 02A0, mask: 0780
```

What is weird is that in the success case, when the mask is logged it is not reset to 0500 as it should. Something really funky is going on, almost as if... the interrupt handler was executed on both cores at once and there was something funny regarding the atomics?

However, I do not understand why I don't see the log from the interrupt in the error case, though the mask register seems to suggest that the interrupt has executed. In the success case the situation is exactly the opposite, the mask is not cleared, but the  log from interrupt is there and the atomic state is correct.

Idea for a potential fix - do not use the atomic variable as communication media, instead use the uart status register directly. The only reason for the interrupt would be to support the waker.

enbyted, 2024-05-04:
I found the issue!

I'm not actually sure if this is a bug in embassy or am I using embassy wrongly.

The issue is caused by my setup - the task reading from uart is running in interrupt executor with priority above the uart's interrupt priority.
Meaning, that when the DMA interrupt is triggered the uart's interrupt doesn't have a chance to run. After moving the task to a normal executor we can see the following log sequence every time:
```
000367.757975 [T embassy_rp::uart] Read starting, rx_errs: 0000, status reg: 0020, mask: 0780
000368.058917 [I embassy_rp::uart] Uart interrupts, errs: 02A0
000368.059078 [T embassy_rp::uart] Read finished, errors: 02A0, rx_errs: 0000, status reg: 02A0, mask: 0500
```

Which is exactly as I expect.
With all that said, getting rid of the atomic variable and using the interrupt simply as waker caller works just as well for this particular issue.

Now, the question is @jamesmunns - is this usecase, of having the executor running at priority higher than most interrupts, supported and it is a bug in embassy, or not?
If yes - I'm happy to create a proper patch to remove the `rx_errs` atomic variable.
If no - this issue can be closed.

jamesmunns, 2024-05-04:
@enbyted nice find! CC @Dirbaio for a policy question

rneswold, 2025-12-03:
Lurker, here.

Did this fix make it into the latest RP support? Or is there more discussion needed?