embassy-rs/embassy #2460: Default `SupplyConfig` for stm32h family?

sgasse, 2024-01-18:
Hi!

We have a few [STM32H745XI discovery boards](https://www.st.com/en/evaluation-tools/stm32h745i-disco.html) used throughout smaller projects. After updating from an older version of `embassy` to the latest version, we ran into a curious issue that is probably related to a change in the default supply config:

The issue happens when we flash a board with the latest version of `embassy`.
- While the board remains connected via USB, every example still works.
- Once disconnected (powered down) and reconnected, none of the examples work (even after reflashing).
- After flashing an example with an old version of `embassy` and reconnecting USB (power cycle), everything works again.

My colleague @KloolK debugged it a while back and ran into the same issue and solution as described in #2430. So when setting the [`SupplyConfig`](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/rcc/h.rs#L126) to `DirectSMPS`, we prevent the board from getting stuck after the next power cycle:

```rust
    let mut config = embassy_stm32::Config::default();
    {
        use embassy_stm32::rcc::*;
        config.rcc.supply_config = SupplyConfig::DirectSMPS;
    }
    let p = embassy_stm32::init(config);
```

Right now, if anyone uses a STM32H745XI discovery board or similar with any of the examples, they may run into the tricky situation of everything working seemingly fine one day, but after a power cycle not being able to run any example. My suggestion is to switch the default `SupplyConfig` to `DirectSMPS` at least for the feature `stm32h745xi-cm7`, maybe also `stm32h745xi-cm4`.

I'd be happy to create a PR but I wanted to make sure that there is no other reason for having the default as it is.
@badrbouslikhin could you comment since you were the last one to edit `SupplyConfig`?

badrbouslikhin, 2024-01-18:
Hi @sgasse,

Just wanted to point you towards the discussion about the SupplyConfig default setting in this PR where they first introduced power supply options: https://github.com/embassy-rs/embassy/pull/2237#discussion_r1412683019. There’s a notable comment there that really sums up why `SupplyConfig::Default’ seemed like the right choice initially:

> When I connected the BOOT0 pin to VDD, my STM32H746XI (MCU) started in System memory.
The internal bootloader automatically sets the PWR_CR3 register to 0x05000046 which corresponds to the default power supply configuration (SupplyConfig::Default).
This configuration ensures that the MCU will boot properly regardless of the hardware setup.
I think it's indeed the most sensible default setting.

However, what you and @wagcampbell have experienced seems to challenge this perspective.

I'm not completely convinced that `DirectSMPS’ is the best default, though. According to the Reference Manual for STM32H745XI:

> The SMPS step-down converter isn’t available on all packages.

It worked for your [STM32H745XI discovery board](https://www.st.com/en/evaluation-tools/stm32h745i-disco.html), but may not be the same for other boards.

I am not sure what would be a better way to handle this besides better documentation.

sgasse, 2024-01-23:
Ah alright, thank you for providing some perspective @badrbouslikhin :pray: I had hoped that the power supply circuitry would be comparable enough for one class of an MCU. Then I agree with you that the only thing to improve the situation may be better documentation.

Would it be worth adding a small section "Troubleshooting" or something in the [README here](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/README.md)?

Not sure though if this is really the place where people would look first when the board stops working. The fact that it happens only after a power cycle and is not reset by a flash memory wipe with STM32CubeProgrammer gave me personally the impression that I either bricked the board or that something about my laptop setup was wrong ^^