embassy-rs/embassy #2464: embassy-rp: UART flush isn't flushy enough

jamesmunns, 2024-01-19:
The blocking uart's blocking+async flush only waits until the buffered data is enqueued, NOT until the data is all out on the wire:

https://github.com/embassy-rs/embassy/blob/871d82de7165781ca6164de72664c9a6a90b3e66/embassy-rp/src/uart/buffered.rs#L419-L427

Instead, you also need to busy loop on the `uart.busy()` flag that reports whether the transmitter is active. I do that here:

https://github.com/jamesmunns/erdnuss-pub/blob/3d98a55aa7a90141a525e4443c927400f8516dd2/source/rp2040/src/lib.rs#L74-L77

There is no interrupt for this state. AFAICT, you have to busy poll from the time of interrupt (when DMA has completed and all bytes enqueued), maybe past the interrupt when the FIFO falls low again, and then busy loop until the transmitter is idle.

This is not compliant with `embedded-io`'s `flush` guarantees. For blocking, this is NBD, we can just busy loop. It's annoying for async tho.

jamesmunns, 2024-01-19:
This is the same kind of deficiency being fixed by #2416

enbyted, 2024-01-19:
The busy loop is exactly what I’m doing for controlling the RS485 transceiver.
However, it would be nice to at least have an async flush for the fifo, then you only have to busy-loop on the last byte.

I guess one could start a (hardware) timer in the uart interrupt for TX fifo empty, that will expire as the last bit is expected to leave the shift register.

enbyted, 2024-01-19:
I re-checked the datasheet and I remembered why in another project I disabled the FIFO for uart - there is no interrupt for an empty FIFO. The best you can get is an interrupt when 4 bytes are remaining to be sent (with watermark level set to 7/8.

That is still better than nothing I suppose - we can have async for everything, but last 5 bytes (4 in the fifo + 1 in the shift register). 

We could get better by disabling the FIFO, but then we get a much higher risk of losing bytes on reception, since there are no separate controls for RX and TX FIFO. In the end, this is using the same interrupts and registers, so it could be a config option with no difference in code for different modes.

//offtopic: I remember scratching my head about how could they release a microcontroller with no hardware TE control and no TX complete style interrupt. I’ve only seen this on some PIC16 micros, never in a 32-bit chip. The best you can do is utilize PIO to control TE for you, but that is hard to make bit-accurate (i.e. make the TE rise right after stop bit).

jamesmunns, 2024-01-19:
@enbyted yeah, there's also a detail I got bit with on my RS485 project: there is a "Line Idle" interrupt, but it only fires if the FIFO is not empty, which means if you use DMA then the FIFO is almost always empty.

That's why I ended up using a line break in my wire protocol instead of just a line idle time.

It's a standard Arm Prime Cell UART (so we can't really fully blame them, other than their choice of IP blocks), but it is a shame it is so meh.

enbyted, 2024-01-19:
That’s also why I’m going with line break, though realistically for my needs 9-bit frames (i.e. what Atmel I think calls the uart with addressing mode or something like that) would be more efficient due to usually sending very short frames. Once I get some other parts of my system running I might work on some support for that, actually.

Anyhow, for this issue fixing the flush to also busy loop until transmission is fully complete sounds like the right approach.
Would you be interested in getting a PIO-based async flush and/or TE controller into embassy? Since PIO can send interrupts it could wake up an async task for flush implementation, plus it’s nearly the same operation as controlling the TE line, I think that would be doable in a relatively-generic way. I’d need to look closer at the APIs, but I might be interested in working on that in near-ish future (a few weeks time).

jamesmunns, 2024-02-23:
Just for the record:

the only solutions I can think of are:

- use a timer and "guess/approximate" how long it'll take to finish flushing
    - CON: This is the most complex and makes `embassy-time` a hard dependency
- do a`while uart.busy() { yield_now().await }` semi busyloop
- do a`while uart.busy() {}` busyloop

marcemmers, 2025-04-25:
@jamesmunns I saw you already started a PR for this a while back, I could pick that up if you want as I ran into this issue as well.

Regarding your solutions, I think the solutions using a timer is still valid? We could at least wait for the up to 4 bytes in the fifo before we do the yielding loop. `Timer::after_us` is already used in the uarts anyway.

jamesmunns, 2025-04-25:
@marcemmers feel free to pick it up! I haven't touched that project since, and I think at the time there wasn't a hard e-time dep already? If that has changed, it seems reasonable to me to use that.

One note is that the timing can get a bit *weird* here at higher baudrates, the rp2040 can do something like 7.8MHz max for UART sending (what I was doing with RS-485), which only takes ~5.1us, probably not worth yielding at all for (<1000 cpu cycles, close to the time I expect to yield and schedule back, even with no other tasks). Not sure where you draw the line though.

jamesmunns, 2025-04-25:
IMO it might be worth adding a configurable for "do spin" and "don't spin, yield/sleep", maybe always defaulting to "don't spin, yield/sleep", so people like me who have their UARTs up at max speed can choose to spin for deterministic timing, in cases like RS-485 where it matters, but most users at normal 9600/115200 speeds aren't going to be in trouble either.