embassy-rs/embassy #2477: Usb composite descriptor can't be setted

Neomaic, 2024-01-25:
Hi,
I'm trying to build a composite devices which have one CDC and one HID,and I noticed a issue [https://github.com/embassy-rs/embassy/issues/1841](url).I tried to use the method in Issue to create but when the MCU is plugged into Ubuntu the dmesg displays can't set config #1, error -32.Is there something wrong with me?
Here is my code

```rust
 let mut config = Config::default();
    config.rcc.hsi48 = true;
    config.rcc.sys_ck = Some(Hertz(48_000_000));
    //config.rcc.pclk = true;
    let mut p = embassy_stm32::init(config);

    info!("Hello World!");

    // Create the driver, from the HAL.
    let driver = Driver::new(p.USB, Irqs, p.PA12, p.PA11);

    // Create embassy-usb Config
    let mut config = embassy_usb::Config::new(0xc0de, 0xcafe);
    config.manufacturer = Some("Topicon");
    config.product = Some("Cradle");
    config.serial_number = Some("12345678");
    config.max_power = 100;
    config.max_packet_size_0 = 64;

    // Create embassy-usb DeviceBuilder using the driver and config.
    // It needs some buffers for building the descriptors.
    let mut device_descriptor = [0; 256];
    let mut config_descriptor = [0; 256];
    let mut bos_descriptor = [0; 256];
    let mut control_buf = [0; 7];
    let request_handler = MyRequestHandler {};

    let mut cdc_state = State::new();
    let mut hid_state = HidState::new();

    let mut builder = Builder::new(
        driver,
        config,
        &mut device_descriptor,
        &mut config_descriptor,
        &mut bos_descriptor,
        &mut [], // no msos descriptors
        &mut control_buf,
    );

    let config = embassy_usb::class::hid::Config {
        report_descriptor: MouseReport::desc(),
        request_handler: Some(&request_handler),
        poll_ms: 60,
        max_packet_size: 8,
    };

    // Create classes on the builder.
    let mut serial_class = CdcAcmClass::new(&mut builder, &mut cdc_state, 64);
    let mut hid_writer = HidWriter::<_, 5>::new(&mut builder, &mut hid_state, config);

    // Build the builder.
    let mut usb = builder.build();

    // Run the USB device.
    let usb_fut = usb.run();
```

Dirbaio, 2024-01-25:
- you may need to add this to get the host to re-enumerate you depending on the board https://github.com/embassy-rs/embassy/blob/main/examples/stm32f1/src/bin/usb_serial.rs#L31-L38
- which stm32 chip are you using, and which board?
- please post full trace-level logs

Neomaic, 2024-01-26:
Hi, I use the stm32f072c8t6 chip and here are the logs
```
0.000000 TRACE BDCR ok: 00008218
└─ embassy_stm32::rcc::bd::{impl#3}::init @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.000000 DEBUG rcc: Clocks { sys: Hertz(48000000), pclk1: Hertz(48000000), pclk1_tim: Hertz(48000000), pclk2: Hertz(48000000), pclk2_tim: Hertz(48000000), hclk1: Hertz(48000000), rtc: Some(Hertz(40000)) }
└─ embassy_stm32::rcc::set_freqs @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:130 
0.000000 INFO  Hello World!
└─ usb_composite::____embassy_main_task::{async_fn#0} @ src/main.rs:32  
0.350372 TRACE allocating type=Interrupt mps=8 interval_ms=255, dir=In
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.350494 TRACE   index=1 addr=64 len=8
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.350616 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=Out
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.350769 TRACE   len_bits = 8400
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.350830 TRACE   index=2 addr=72 len=64
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.350952 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=In
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.351104 TRACE   index=2 addr=136 len=64
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.351257 TRACE allocating type=Interrupt mps=8 interval_ms=60, dir=In
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.351379 TRACE   index=3 addr=200 len=8
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.351531 INFO  USB: device_descriptor used: 18
└─ embassy_usb::builder::{impl#1}::build @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:143 
0.351593 INFO  USB: config_descriptor used: 103
└─ embassy_usb::builder::{impl#1}::build @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:143 
0.351684 INFO  USB: bos_descriptor used: 12
└─ embassy_usb::builder::{impl#1}::build @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:143 
0.351776 INFO  USB: msos_descriptor used: 0
└─ embassy_usb::builder::{impl#1}::build @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:143 
0.351837 INFO  USB: control_buf size: 7
└─ embassy_usb::builder::{impl#1}::build @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:143 
0.351928 TRACE allocating type=Control mps=64 interval_ms=0, dir=Out
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.352050 TRACE   len_bits = 8400
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.352111 TRACE   index=0 addr=208 len=64
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.352233 TRACE allocating type=Control mps=64 interval_ms=0, dir=In
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.352355 TRACE   index=0 addr=272 len=64
└─ embassy_stm32::usb::usb::{impl#2}::alloc_endpoint @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.352478 TRACE enabled
└─ embassy_stm32::usb::usb::{impl#3}::start @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.352569 TRACE usb: power detected
└─ embassy_usb::{impl#2}::handle_bus_event::{async_fn#0} @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:117 
0.352630 TRACE RESET
└─ embassy_stm32::usb::usb::{impl#4}::poll::{async_fn#0}::{closure#0} @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.352691 TRACE usb: reset
└─ embassy_usb::{impl#2}::handle_bus_event::{async_fn#0} @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:117 
0.352783 TRACE SETUP read waiting
└─ embassy_stm32::usb::usb::{impl#12}::setup::{async_fn#0} @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.352844 TRACE wait_enabled OUT WAITING
└─ embassy_stm32::usb::usb::{impl#9}::wait_enabled::{async_fn#0} @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.355529 TRACE usb: resume
└─ embassy_usb::{impl#2}::handle_bus_event::{async_fn#0} @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:117 
0.355590 TRACE usb: suspend
└─ embassy_usb::{impl#2}::handle_bus_event::{async_fn#0} @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:117 
0.724670 TRACE RESET
└─ embassy_stm32::usb::usb::{impl#4}::poll::{async_fn#0}::{closure#0} @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.724761 TRACE usb: reset
└─ embassy_usb::{impl#2}::handle_bus_event::{async_fn#0} @ /home/neo/work/mcu/embassy/embassy-usb/src/fmt.rs:117 
0.724853 TRACE SETUP read waiting
└─ embassy_stm32::usb::usb::{impl#12}::setup::{async_fn#0} @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
0.804504 TRACE READ DONE, rx_len = 8
└─ embassy_stm32::usb::usb::{impl#7}::read_data @ /home/neo/work/mcu/embassy/embassy-stm32/src/fmt.rs:117 
[... 152 lines cut ...]
0.931488 TRACE control: data_in
[... cut ...]
└─ panic_probe::print_defmt::print @ /home/neo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:104 
``` 

The log has clearly shows that the control buffer too small.
But here's the strange thing.When I increase the control buffer, the usb is recognized normally, and when I change it back to the original value, it is also recognized normally.
It is possible that the computer has been enumerated as VCP by default because it was previously tested with a separate CDC project.
So it looks like it's been resolved.And i will close this issue

closed 2024-01-26 by Neomaic