embassy-rs/embassy #2479: stm32c0 ADC support

usbalbin, 2024-01-25:
Hi! :)

Are there any plans for stm32c0 ADC support?

usbalbin, 2024-01-28:
If this is something you want, do you have any pointers for where to begin? 

I have little to no experience with embassy, but a little bit of experience with stm32-rs/stm32g4xx-hal

Dirbaio, 2024-01-28:
First step is to add the registers to [stm32-data](https://github.com/embassy-rs/stm32-data). See the ADC peripherals for C0 [have no `registers` field](https://github.com/embassy-rs/stm32-data-generated/blob/main/data/chips/STM32C031K6.json#L42) (compare it [with G0](https://github.com/embassy-rs/stm32-data-generated/blob/main/data/chips/STM32G031K8.json#L219) which does). 

follow the steps in the readme to extract ADC register yamls from all the SVDs. Then diff them between themselves. If you're lucky you'll find the ADC registers in C0 are identical (after cleaning up!) to another family that we already have registers for, in which case you can add an entry mapping the C0 version to that version. If not you'll have to add a new version.

Second step is adding support in embassy-stm32. [Here's all the versions](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/adc/mod.rs#L7-L14) mapped to the driver they're using. Again, hopefully it's similar enough to an existing driver that you can reuse it (pointing both families to the same .rs file). If not you'll have to add a new one.

usbalbin, 2024-01-30:
Thank you!

@Dirbaio 
> First step is to add the registers to [stm32-data](https://github.com/embassy-rs/stm32-data). See the ADC peripherals for C0 [have no `registers` [...]

First off `tmp/c031.yaml` and `tmp/c011.yaml` are identical. However `diff tmp/g031.yaml tmp/c031.yaml` shows very few, yet some differences from what I can tell:

* Different phrasing and or formatting in description fields
* c031.yaml does not seem to have even a single enum
* c031 is missing the bit 24 VBATEN in ADC CCR which [RM0490](https://www.st.com/resource/en/reference_manual/rm0490-stm32c0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) agrees that it should not have.
* The analog watchdog config registers have bits for 4 extra adc channels(19-22)
* Differences in how the bits in CHSELR0 are divided into fields
  * g031 has one 19 bits wide field
  * c031 has 23 single bit fields. From what I can tell c031 better describes what g0x1's [RM says](https://www.st.com/resource/en/reference_manual/rm0444-stm32g0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf#page=395) but with 4 extra fields for the extra channels (19-22)

> # Adding support for a new peripheral
> [...]
> * Identify how many actually-different (incompatible) versions of the peripheral exist, as **we must not merge them**. Name them v1, v2.. (if possible, by order of chip release date, see [here](https://docs.google.com/spreadsheets/d/1-R-AjYrMLL2_623G-AFN2A9THMf8FFMpFD4Kq-owPmI/edit#gid=1972450814).

Is this close enough that I should merge c031's extra fields for the 4 ADC channels into `data/registers/adc_g0.yaml` and presumably rename that to `data/registers/adc_v5.yaml`? If so, should I do anything to ensure those few differences that are, are enforced somehow?

shufps, 2024-02-04:
I have a similar issue with ADC on my L052 - I tracked it down to `aditf4_v2_0` version that should work fine with `v1` driver but when adding it to the periph-mapping and rebuilding the PAC I still have no ADC :thinking:

chudsaviet, 2025-02-04:
@usbalbin , did you end up adding C0 ADC support?

usbalbin, 2025-02-04:
Sorry, no I did not get around to it

chudsaviet, 2025-02-05:
I have started working on it, 'stm32-data' change is in https://github.com/chudsaviet/stm32-data/blob/c0_adc/data/registers/adc_c0.yaml.

@Dirbaio , I see somewhat messy code structure in  embassy-stm32/src/adc. Shall I add a 'c0.rs' or add C0 support to 'v3.rs' or 'v4.rs'?

Dirbaio, 2025-02-06:
If the c0 registers are similar to v3/v4 you could try extending those to support c0, otherwise adding c0.rs is fine.

Yes, it's messy since it's grown "organically" over the years. Ideally it should be cleaned up and deduplicated.

chudsaviet, 2025-02-06:
Its ok, mess happens. 
Whats the difference between v1-v4? Are v1-v3 just legacy, or versions for specific MCUs? Shall I try, for example, integrate into v3 instead of v4?

Dirbaio, 2025-02-07:
they refer to different versions of the ADC peripheral found in different STM32 families. See this https://github.com/embassy-rs/stm32-data/blob/6dbba9fcfebd2af002b380eed0a63ea16ca1e34e/stm32-data-gen/src/perimap.rs#L92-L127

chudsaviet, 2025-02-10:
@Dirbaio , I created a separate version of C0 ADC peripheral, because it looks different enough - https://github.com/chudsaviet/embassy/blob/c0/embassy-stm32/src/adc/c0.rs (don't be disgusted by the code and commit messages, I will clean it before doing pull request).
Added an `stm32-data` entry of course - https://github.com/chudsaviet/stm32-data/blob/c0_adc/data/registers/adc_c0.yaml

I'm struggling to figure out code generation when in comes to a peripheral implementing `RccPeripheral` trait. The C0 ADC just do not implement it. If I'm correct, if ADC clock is not enabled in RCC - it does not start, waits for `ADRDY` bit forever. 

Could you please point me to a place in code where I can make C0 ADC peripheral implement `RccPeripheral` trait?

chudsaviet, 2025-02-12:
Nevermind - I figured it out.

chudsaviet, 2025-02-17:
Here we go, first merge request: https://github.com/embassy-rs/stm32-data/pull/569.

chudsaviet, 2025-02-18:
Here we go again:
https://github.com/embassy-rs/stm32-data/pull/570
https://github.com/embassy-rs/embassy/pull/3899