embassy-rs/embassy #2489: RP-2040 PicoW support for bonded encrypted lvl 4 BLE and HTTPS TLS on wifi?

mytechnotalent, 2024-01-28:
Really excited about the embassy-rs project.  I do not see any examples for BLE and more specifically BLE for bonded level 4 encryption nor do I see any HTTPS examples with TLS.  Are these going to at some point be part of the RP-2040 port?

Dirbaio, 2024-01-28:
you can do HTTPS with [reqwless](https://github.com/drogue-iot/reqwless) and [embedded-tls](https://github.com/drogue-iot/embedded-tls). 

Bluetooth is WIP. There's HCI support here https://github.com/embassy-rs/embassy/pull/1820 but it's going to take a while until there's a Rust bluetooth host stack on top of it capable of the fancier things like bonding.

mytechnotalent, 2024-01-29:
Thank you @Dirbaio for taking the time to respond!  

Is reqwless and embedded-tls compatiable with Embassy?

Regarding BLE thank you for giving me a heads-up that it might be a while.