embassy-rs/embassy #2494: Extract sysnopsys part of embassy-stm32/usb_otg into it's own crate

Dominaezzz, 2024-01-29:
The `usb_otg` code in https://github.com/embassy-rs/embassy/tree/main/embassy-stm32/src/usb_otg can be used for other MCUs besides the STM32, i.e. the ESP32-S2 and the ESP32-S3. (This was the result of #2488)

There are a couple decisions and design choices that need making. I also have questions as someone who is new to USB and PACs.

1. What should this new crate be called? `embassy-usb-synopsys`? Maybe throw `otg` or `driver` in there? idk.
2. STM32 and ESP32 have different register offsets for the peripheral, how should this be handled? Should there be a trait for exposing each register or should the crate have `#[cfg(..)]`s for each mcu? The latter seems like the way.
3. Is it alright to make breaking changes? I don't think there's a choice in the matter.

Some initial question after a scan.

Where is [`crate::pac::otg::Otg`](https://github.com/embassy-rs/embassy/blob/3cd0a54e0e3e7dd4b480034106b8b24144d2385b/embassy-stm32/src/usb_otg/mod.rs#L18) defined?

Any reason [this](https://github.com/embassy-rs/embassy/blob/702007f68b31718d0bea5af403c4c0a299fb6b6e/embassy-stm32/src/usb_otg/usb.rs#L12) isn't `use core::future::poll_fn`?

Why is there a difference in register access between esp32 and stm32? Isn't it all svd2rust?
STM32 is like this.
```Rust
T::regs().gintmsk().write(|w| {
    w.set_usbrst(true);
    w.set_enumdnem(true);
    w.set_usbsuspm(true);
    w.set_wuim(true);
    w.set_iepint(true);
    w.set_oepint(true);
    w.set_rxflvlm(true);
    w.set_srqim(true);
    w.set_otgint(true);
});
```
ESP32 is like this.
```Rust
T::regs().gintmsk().write(|w| {
    w.usbrstmsk().set_bit();
    w.enumdonemsk().set_bit();
    w.usbsuspmsk().set_bit();
    w.wkupintmsk().set_bit();
    w.iepintmsk().set_bit();
    w.oepintmsk().set_bit();
    w.rxflvimsk().set_bit();
    w.sessreqintmsk().set_bit();
    w.otgintmsk().set_bit();
    w
});
```

What is [`<T as RccPeripheral>::enable_and_reset();`](https://github.com/embassy-rs/embassy/blob/702007f68b31718d0bea5af403c4c0a299fb6b6e/embassy-stm32/src/usb_otg/usb.rs#L657) about? This won't be accessible from the synopsys crate so I'll have to put it behind a trait.

Not sure what to do with [this](https://github.com/embassy-rs/embassy/blob/702007f68b31718d0bea5af403c4c0a299fb6b6e/embassy-stm32/src/usb_otg/usb.rs#L659C12-L659C21), I guess that'll go behind a trait too.

[`regs::Fifo`](https://github.com/embassy-rs/embassy/blob/702007f68b31718d0bea5af403c4c0a299fb6b6e/embassy-stm32/src/usb_otg/usb.rs#L1303)?

Besides the above, I think the rest is just a matter of taking a snapshot of the OTG part of the PAC and using it in the crate and expose some of the above via some traits I guess.

Dominaezzz, 2024-01-30:
> Where is [`crate::pac::otg::Otg`](https://github.com/embassy-rs/embassy/blob/3cd0a54e0e3e7dd4b480034106b8b24144d2385b/embassy-stm32/src/usb_otg/mod.rs#L18) defined?

https://github.com/embassy-rs/stm32-data-generated/blob/main/stm32-metapac/src/peripherals/otg_v1.rs

MabezDev, 2024-02-20:
The reason for the difference is that esp-hal is using svd2rust pacs, where as embassy-stm32 is using [chip tool](https://github.com/embassy-rs/chiptool) pacs. Upstream in usb_otg they used [ral](https://github.com/adamgreig/stm32ral) to generate a "mini" pac for just the USB registers.

I'm not sure what's the best approach here.

Dominaezzz, 2024-02-21:
@Dirbaio I'm guessing you have a preference for chip tool?

PR #2871 (2024-04-26): Extract Synopsys USB OTG driver from embassy-stm32
Closes #2494
closed 2024-04-26 by Dirbaio