embassy-rs/embassy #2496: Cannot receive IPv4 multicast packets on STM32H7

mcbridejc, 2024-01-30:
## Relevant Versions

embassy-net  = 0.4.0
embassy-stm32 = 0.1.0

chip = stm32h725zg

## Observation

UDP packets destined to multicast address 224.0.0.1 or another 224.0.0.x address registered via [`join_multicast_group`](https://docs.embassy.dev/embassy-net/git/default/struct.Stack.html#method.join_multicast_group) are never received on a socket bound to the local IP address and correct port. 

## Cause

The MAC packet filter is blocking them, because the MAC does not pass multicast packets by default. 

A quick solution which gets packets through is to toggle the PR bit, which enables all multicast address to get through the filter.

```rust
let mac = embassy_stm32::pac::ETH;
mac.ethernet_mac().macpfr().modify(|w| w.set_pr(true));
```

A more targetted solution would be to add addresses to the 4 address filter registers, although this limits the number of multicast groups which can be joined (seems fine to me, it's limited by a fixed storage table which defaults to 4 anyway in smoltcp). 

I'm up to putting a PR together to address this, but interested to hear if there are opinions on direction of fix.

itmmckernan, 2024-12-27:
This issue is also present in the V1 stm32 Ethernet driver, adding the code below to the new function in `embassy-stm32/eth/v1/mod.rs` fixed it for me on the STM32F429IIT6.
`mac.macffr().modify(|w| {w.set_pam(true); });`

Dirbaio, 2024-12-27:
@itmmckernan can you send a PR for v1?

also, this was fixed for v2 in https://github.com/embassy-rs/embassy/pull/3488

closed 2025-01-02 by Dirbaio