embassy-rs/embassy #2498: Cannot use i2c with stm32f030k6

tomhampshire, 2024-01-30:
With the dependency: `stm32f030k6` of `embassy-stm32` version 0.1.0, it is impossible to configure the interrupts for use with i2c. [Here is a repo](https://github.com/tomhampshire/stm32f030k6/blob/2b0f98112f6140519109ccc2245e3090873297d4/) which can be pulled and which demonstrates the problem. I will reference the code below. Note that the issues are listed in the comments.
main.rs:
```rust
#![no_std]
#![no_main]

use defmt_rtt as _;
use embassy_executor::Spawner;
use embassy_stm32::dma::NoDma;
use embassy_stm32::time::Hertz;
use embassy_stm32::{bind_interrupts, i2c, peripherals};
use panic_probe as _;

// Options for bind_interrupts are listed below:

// This fails as: cannot find type `InterruptHandler` in module `i2c`
bind_interrupts!(struct Irqs {
    I2C1 => i2c::InterruptHandler<peripherals::I2C1>;
});

// This fails as:
// cannot find type `I2C1_EV` in module `$crate::interrupt::typelevel`
// not found in `$crate::interrupt::typelevel` [E0412]
// and:
// cannot find type `I2C1_ER` in module `$crate::interrupt::typelevel`
// not found in `$crate::interrupt::typelevel` [E0412]

bind_interrupts!(struct Irqs {
    I2C1_EV => i2c::EventInterruptHandler<peripherals::I2C1>;
    I2C1_ER => i2c::ErrorInterruptHandler<peripherals::I2C1>;
});

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let mut config = embassy_stm32::Config::default();
    let p = embassy_stm32::init(config);

    let i2c_config = i2c::Config::default();

    let mut i2c1 = i2c::I2c::new(
        p.I2C1,
        p.PA9,
        p.PA10,
        Irqs,
        NoDma,
        NoDma,
        Hertz(1_000_000),
        i2c_config,
    );

    loop {}
}
```

.cargo/config.toml:
```toml
[target.thumbv6m-none-eabi]
runner = 'probe-rs run --chip STM32F030K6TX'

[build]
target = "thumbv6m-none-eabi"

[env]
DEFMT_LOG = "trace"
```

Cargo.toml:
```toml
[package]
name = "power-supply"
version = "0.1.0"
edition = "2021"
# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
embassy-stm32 = { version = "0.1.0", features = [
  "defmt",
  "rt",
  "memory-x",
  "stm32f030k6",
  "time-driver-any",
  "exti",
  "unstable-pac",
] }
cortex-m = { version = "0.7.6", features = [
  "inline-asm",
  "critical-section-single-core",
] }
cortex-m-rt = "0.7.0"
defmt = "0.3"
defmt-rtt = "0.4"
panic-probe = "0.3"
embassy-sync = { version = "0.5.0", features = ["defmt"] }
embassy-executor = { version = "0.5.0", features = [
  "arch-cortex-m",
  "executor-thread",
  "executor-interrupt",
  "defmt",
  "integrated-timers",
] }
embassy-time = { version = "0.3.0", features = [
  "defmt",
  "defmt-timestamp-uptime",
  "tick-hz-32_768",
] }
static_cell = "2"
portable-atomic = { version = "1.5", features = ["unsafe-assume-single-core"] }

[profile.release]
debug = 2
```

jamesmunns, 2024-01-30:
This is likely a dupe of https://github.com/embassy-rs/embassy/issues/2253

tomhampshire, 2024-01-30:
It is, and was fixed with:
```rust
bind_interrupts!(struct Irqs {
    I2C1 => i2c::EventInterruptHandler<peripherals::I2C1>, i2c::ErrorInterruptHandler<peripherals::I2C1>;
});
```
Thanks for your help!

closed 2024-01-30 by jamesmunns