embassy-rs/embassy #2515: stm32: RCC refactor tracking issue

Dirbaio, 2024-02-01:
## The 3 designs

Note RCC in embassy-stm32 is a bit "in flux". Currently we have three API designs for RCC depending on the family, for historical reasons. Some have been ported from stm32-rs, some have been developed for embassy but before we knew what the best design is.

- v1: user specifies the target frequencies, RCC tries to solve for the best config (prescalers etc) to achieve that. [example](https://github.com/embassy-rs/embassy/blob/c4839e4671ade4b69f47e5436b5e44fae9109159/embassy-stm32/src/rcc/f1.rs#L19-L23)
- v2: user specifies prescalers/dividers instead of target freqs. Sysclk mux is an enum whose variants contain the config for the picked clock source. [example](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/rcc/g0.rs#L21-L28)
- v3: user specifies prescalers/dividers. There's separate fields for each clock source (HSE, HSI, MSI, PLL..) to enable and configure them. Sysclk mux is an enum without data that just chooses the source (which the user must ensure is enabled, of course) [example](https://github.com/embassy-rs/embassy/blob/c4839e4671ade4b69f47e5436b5e44fae9109159/embassy-stm32/src/rcc/l.rs#L39-L54)

## Rationale for the v3 design

- Directly specifying frequencies (v1 api design) turned out to be quite troublesome. Solving for the best RCC config backwards from the target frequencies is hard, requires a lot of code that the compiler isn't always able to optimize out, and it's often impossible to exactly match the requested freqs, so we have to pick the closest one making it quite unpredictable for the end user.
- The v2 design rules out some configurations that the hardware allows. For example you might want to run sysclk from PLL from HSE, but run some peripheral from HSI (with a CCIPR register). The hardware does allow having HSE and HSI on at the same time, but you can't do it if the mux enum forces you to pick. With the v3 design you can enable all of HSI, HSE, PLL just fine, then pick one for sysclk and a different one for a particular peripheral.

## The refactor

We should port all RCCs to the v3 design, both for consistency between chips and to get the v3 benefits.

- [ ] STM32F1: v1
- [x] STM32F2: v3
- [x] STM32L1: v3
- [x] STM32F4: v3
- [ ] STM32F0: v1
- [ ] STM32F3: v1
- [x] STM32F7: v3
- [x] STM32L0: v3
- [x] STM32L4: v3
- [x] STM32H7: v3
- [ ] STM32G0: v2
- [x] STM32L5: v3
- [ ] STM32G4: v2
- [x] STM32WB: v3
- [x] STM32WL: v3
- [ ] STM32U5: v2
- [ ] STM32C0: v2
- [x] STM32H5: v3
- [ ] STM32WBA: v2

While we're at it, it'd be great to unify similar families into a single file. I have already unified a few into `l.rs`, `h.rs`, `f.rs` , but I think the remaining ones are different enough that we might want to keep them separate. 

Some candidates to unify are:

- F3, F0
- G0, C0 (it's a cut-down G0), and also maybe G4
- U5, WBA (?)

barnabywalters, 2024-02-13:
I made a start on G4, not quite ready to PR yet but hopefully in the next days.

Dirbaio, 2024-03-03:
this is now DONE, finally :partying_face:

closed 2024-03-03 by Dirbaio