embassy-rs/embassy #2522: STM32 gen_waveform on TIM2 not working

Palm5, 2024-02-02:
Using stm32 f401re.
Tried to set up PWM output for TIM2 but waveform is not generating correctly.
While analyzing with a logic analizer I can see that the output is switching from low to high whenever either TIM1 or TIM3 outputs are being toggled.
Example of what i'm trying to do:

```
let p = embassy_stm32::init(Default::default());
let pwm = SimplePwm::new(
      p.TIM2,
      None,
      None,
      Some(PwmPin::new_ch3(p.PB10, OutputType::PushPull)),
      None,
      FREQUENCY,
      Default::default(),
    );

let pulses = [pwm.get_max_duty() / 2; 255];
pwm.gen_waveform(p.DMA1_CH1, embassy_stm32::timer::Channel::Ch3, &pulses).await;
```

I'm using "time-driver-tim5" feature of embassy-stm32 crate to make TIM2 available, since 'time-driver-any' would use it.
Unfortunately the MCU has only 4 timers with DMAs and i need all 4 for my use case.
With TIM1, TIM3 and TIM4 no problems whatsoever.

CNLHC, 2025-01-01:
The problem is the waveform_xx currently only supports the 16bit timer, if you assign a 32bit timer to it then it will not work. I am going to write a patch to fix it today.

CNLHC, 2025-01-24:
Seems that this issue can be closed

closed 2025-01-24 by Dirbaio