embassy-rs/embassy #2527: error: could not compile 'embassy-executor' (lib) - invalid instructions mnemonic 'sev'

ismet55555, 2024-02-03:
Hello

I'm trying to get my simple blink LED program to compile, but I'm getting the error below. 

**Host:** `Linux my-computer 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`
**Target:** `Raspberry Pi Pico W (RP2040)`

The program itself isn't anything overly complicated:
```rs
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use defmt_rtt as _;
use embassy_executor::Spawner;
use embassy_rp::gpio::{Level, Output};
use embassy_time::{Duration, Timer};
use panic_probe as _;

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    defmt::info!("Initializing ...");

	.....
}

```

**.config/config.toml:**
```toml
[build]
target = "thumbv6m-none-eabi"

[target.thumbv6m-none-eabi]
runner = "probe-rs run --chip=RP2040"
rustflags = ["-C", "link-args=-Tlink.x -Tlink-rp.x -Tdefmt.x"]

[env]
DEFMT_LOG = "debug"
```

**Cargo.toml**
```toml
[package]
edition = "2021"
name = "embedded-rust-1"
readme = "README.md"
version = "0.1.0"

[dependencies]
cortex-m-rt = "0.7.3"
defmt = "0.3"
defmt-rtt = "0.4"

[dependencies.embassy-executor]
git = "https://github.com/embassy-rs/embassy.git"
features = ["task-arena-size-32768", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"]

[dependencies.embassy-rp]
git = "https://github.com/embassy-rs/embassy.git"
features = ["critical-section-impl", "time-driver", "defmt"]

[dependencies.embassy-time]
git = "https://github.com/embassy-rs/embassy.git"
features = ["defmt", "defmt-timestamp-uptime"]

[dependencies.panic-probe]
version = "0.3"
features = ["print-defmt"]
```

**Output:**
```bash
cargo clean && cargo run
     Removed 418 files, 154.9MiB total
   Compiling proc-macro2 v1.0.78
   Compiling unicode-ident v1.0.12
   Compiling syn v1.0.109
   Compiling cfg-if v1.0.0
   Compiling version_check v0.9.4
   Compiling autocfg v1.1.0
   Compiling libc v0.2.153
   Compiling defmt v0.3.5
   Compiling thiserror v1.0.56
   Compiling cortex-m v0.7.7
   Compiling nb v1.1.0
   Compiling defmt-macros v0.3.6
   Compiling semver-parser v0.7.0
   Compiling parking_lot_core v0.9.9
   Compiling memchr v2.7.1
   Compiling semver v0.9.0
   Compiling nb v0.1.3
   Compiling smallvec v1.13.1
   Compiling scopeguard v1.2.0
   Compiling crunchy v0.2.2
   Compiling rustix v0.38.31
   Compiling bitflags v1.3.2
   Compiling void v1.0.2
   Compiling regex-syntax v0.8.2
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling lock_api v0.4.11
   Compiling rustc_version v0.2.3
   Compiling embedded-hal v0.2.7
   Compiling equivalent v1.0.1
   Compiling critical-section v1.1.2
   Compiling bitflags v2.4.2
   Compiling hashbrown v0.14.3
   Compiling quote v1.0.35
   Compiling bare-metal v0.2.5
   Compiling embassy-time-driver v0.1.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling aho-corasick v1.1.2
   Compiling siphasher v0.3.11
   Compiling syn v2.0.48
   Compiling tiny-keccak v2.0.2
   Compiling linux-raw-sys v0.4.13
   Compiling phf_shared v0.10.0
   Compiling log v0.4.20
   Compiling bit-vec v0.6.3
   Compiling futures-core v0.3.30
   Compiling fixedbitset v0.4.2
   Compiling once_cell v1.19.0
   Compiling futures-sink v0.3.30
   Compiling dirs-sys-next v0.1.2
   Compiling indexmap v2.2.2
   Compiling vcell v0.1.3
   Compiling either v1.9.0
   Compiling embassy-time-queue-driver v0.1.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling precomputed-hash v0.1.1
   Compiling dirs-next v2.0.0
   Compiling term v0.7.0
   Compiling parking_lot v0.12.1
   Compiling new_debug_unreachable v1.0.4
   Compiling itertools v0.10.5
   Compiling volatile-register v0.2.2
   Compiling bit-set v0.5.3
   Compiling ena v0.14.2
   Compiling ascii-canvas v3.0.0
   Compiling string_cache v0.8.7
   Compiling litrs v0.4.1
   Compiling regex-syntax v0.6.29
   Compiling paste v1.0.14
   Compiling cortex-m-rt v0.7.3
   Compiling petgraph v0.6.4
   Compiling diff v0.1.13
   Compiling is-terminal v0.4.10
   Compiling heapless v0.8.0
   Compiling bitfield v0.13.2
   Compiling pin-project-lite v0.2.13
   Compiling byteorder v1.5.0
   Compiling futures-task v0.3.30
   Compiling unicode-xid v0.2.4
   Compiling pin-utils v0.1.0
   Compiling hash32 v0.3.1
   Compiling regex-automata v0.4.5
   Compiling document-features v0.2.8
   Compiling strsim v0.10.0
   Compiling debug-helper v0.3.13
   Compiling embedded-io-async v0.6.1
   Compiling embedded-hal-async v1.0.0
   Compiling stable_deref_trait v1.2.0
   Compiling ident_case v1.0.1
   Compiling fnv v1.0.7
   Compiling embedded-hal v1.0.0
   Compiling crc-any v2.4.4
   Compiling num-traits v0.2.17
   Compiling arrayvec v0.7.4
   Compiling embassy-sync v0.5.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling az v1.2.1
   Compiling embedded-io v0.6.1
   Compiling typenum v1.17.0
   Compiling rp2040-boot2 v0.3.0
   Compiling termcolor v1.4.1
   Compiling rp-pac v6.0.0
   Compiling unicode-width v0.1.11
   Compiling fixed v1.24.0
   Compiling embassy-hal-internal v0.1.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling embedded-storage v0.3.1
   Compiling atomic-polyfill v1.0.3
   Compiling embedded-storage-async v0.4.1
   Compiling codespan-reporting v0.11.1
   Compiling futures-channel v0.3.30
   Compiling darling_core v0.20.5
   Compiling half v2.3.1
   Compiling embassy-rp v0.1.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling defmt-rtt v0.4.0
   Compiling panic-probe v0.3.1
   Compiling futures-io v0.3.30
   Compiling embassy-executor v0.5.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling embassy-futures v0.1.1 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling bytemuck v1.14.1
   Compiling embedded-hal-nb v1.0.0
   Compiling rand_core v0.6.4
   Compiling regex v1.10.3
   Compiling lalrpop-util v0.19.12
   Compiling lalrpop v0.19.12
   Compiling thiserror-impl v1.0.56
   Compiling futures-macro v0.3.30
   Compiling num_enum_derive v0.5.11
   Compiling cortex-m-rt-macros v0.7.0
   Compiling darling_macro v0.20.5
   Compiling defmt-parser v0.3.3
   Compiling futures-util v0.3.30
   Compiling darling v0.20.5
   Compiling num_enum v0.5.11
   Compiling embassy-executor-macros v0.4.1 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling pio v0.2.1
   Compiling futures v0.3.30
   Compiling embassy-time v0.3.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling embassy-usb-driver v0.1.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
   Compiling embassy-embedded-hal v0.1.0 (https://github.com/embassy-rs/embassy.git#88e7e1ce)
error: invalid instruction mnemonic 'sev'
  --> /home/me/.cargo/git/checkouts/embassy-9312dcb0ed774b29/88e7e1c/embassy-executor/src/arch/cortex_m.rs:13:31
   |
13 |             core::arch::asm!("sev");
   |                               ^
   |
note: instantiated into assembly here
  --> <inline asm>:2:2
   |
2  |     sev
   |     ^^^

error: could not compile `embassy-executor` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
```

Dirbaio, 2024-02-04:
If the project is part of a Cargo workspace, make sure to add `resolver = 2` to the workspace declaration. 

(I recommend NOT putting embedded projects in workspaces, they tend to create issues with Cargo features like these)

ismet55555, 2024-02-16:
Sorry if i'm not following, this is fairly new to me, I was trying to get a very basic example to compile and run first before going forward.

What do you mean by this?

My project just looks like this, with the above specifics

```txt
.
── assets
│   └── 69-probe-rs.rules
├── Cargo.lock
├── Cargo.toml
├── .config
│   └── config.toml
├── .git
├── .gitignore
├── memory.x
├── README.md
├── rust-toolchain.toml
├── src
│   └── main.rs
└── target
```

Dirbaio, 2024-02-16:
ah sorry, I saw that your `config.toml` was correct (it's setting the correct target), so I assumed it had to be an issue with Cargo workspaces.

Actually the issue is it needs to be in `.cargo/config.toml`, not `.config/config.toml`, I just noticed :sweat_smile:

closed 2024-02-16 by ismet55555