embassy-rs/embassy #2531: no field `hsi48`, `pll1_p` on type Temp

aeblyve, 2024-02-04:
To reproduce:

`cd examples/stm32f0`
Change `stm32f091rc` to `stmf072c8` (keep rest of `Cargo.toml` the same)
`cargo build`

The result is:

```
error[E0609]: no field `hsi48` on type `Temp`
   --> /tmp/blorg/target/thumbv6m-none-eabi/debug/build/embassy-stm32-fc57c2643893afc2/out/_macros.rs:1:347
    |
1   |   ...rcc :: Clocks { hclk1 : all . hclk1 , hsi : all . hsi , hsi48 : all . hsi48 , lse : all . lse , pclk1 : all . pclk1 , pclk1_tim : all . pclk1_tim , pclk2 : all...
    |                                                                            ^^^^^ unknown field
    |
   ::: /home/ablyve/tmp/embassy/embassy-stm32/src/rcc/f0.rs:162:5
    |
162 | /     set_clocks!(
163 | |         hsi: None,
164 | |         lse: None,
165 | |         sys: Some(Hertz(real_sysclk)),
...   |
171 | |         rtc: rtc,
172 | |     );
    | |_____- in this macro invocation
    |
    = note: this error originates in the macro `set_clocks` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0609]: no field `pll1_p` on type `Temp`
   --> /tmp/blorg/target/thumbv6m-none-eabi/debug/build/embassy-stm32-fc57c2643893afc2/out/_macros.rs:1:492
    |
1   |   ..., pclk2 : all . pclk2 , pclk2_tim : all . pclk2_tim , pll1_p : all . pll1_p , rtc : all . rtc , sys : all . sys , }) ; } } ; }#[allow(unused)]
    |                                                                           ^^^^^^ unknown field
    |
   ::: /home/ablyve/tmp/embassy/embassy-stm32/src/rcc/f0.rs:162:5
    |
162 | /     set_clocks!(
163 | |         hsi: None,
164 | |         lse: None,
165 | |         sys: Some(Hertz(real_sysclk)),
...   |
171 | |         rtc: rtc,
172 | |     );
    | |_____- in this macro invocation
    |
    = note: this error originates in the macro `set_clocks` (in Nightly builds, run with -Z macro-backtrace for more info)
```

aeblyve, 2024-02-04:
element chat from @adamgreig submits https://github.com/embassy-rs/embassy/pull/2521 introduced the bug

Ragarnoy, 2024-02-04:
same issue on stm32l431cb for the field pllsai2_p
ie.
```
error[E0609]: no field `pllsai2_p` on type `Temp`
   --> embassy/examples/stm32l4/target/thumbv7em-none-eabi/release/build/embassy-stm32-aab123763c6d6b70/out/_macros.rs:1:595
    |
1   |   ...l1_q , pllsai1_p : all . pllsai1_p , pllsai2_p : all . pllsai2_p , rtc : all . rtc , sai1_extclk : all . sai1_extclk , sys : all . sys...
    |                                                             ^^^^^^^^^ unknown field
    |
   ::: /home/ragarnoy/CLionProjects/embassy/embassy-stm32/src/rcc/l.rs:378:5
    |
378 | /     set_clocks!(
379 | |         sys: Some(sys_clk),
380 | |         hclk1: Some(hclk1),
381 | |         #[cfg(any(stm32l4, stm32l5, stm32wb, stm32wl))]
...   |
423 | |         lse: None,
424 | |     );
    | |_____- in this macro invocation
    |
    = note: this error originates in the macro `set_clocks` (in Nightly builds, run with -Z macro-backtrace for more info)

```

PR #2533 (2024-02-04): stm32/rcc: fix build for some f0 and l4 chips.
Fixes #2531
commit e3fe0842 (2024-02-04): stm32/rcc: fix build for some f0 and l4 chips.

Fixes #2531
closed 2024-02-04 by Dirbaio