embassy-rs/embassy #2536: Multiple Executors on Single Core Chips

logerup, 2024-02-06:
I am currently working with an STM32H743 and would like to integrate multiple different priority levels in my program. 

The approach I was hoping to use is multiple executors, as stated in the embassy documentation "You can also create the Executor manually, and you can in fact create multiple Executors." 
Nowhere does it state that this is only possible for multi-core chips, but in playing around with it and from my understanding on how the executor should work in managing the underlying tasks, it seems like it is only possible to have one executor per core.

I was also not able to find anything in an example, that uses more than one executor either. 

Could anyone elaborate on whether I'm correct or, if not, how to correctly use multiple executors on a single-core chip?

Dirbaio, 2024-02-06:
this is the example you're looking for https://github.com/embassy-rs/embassy/blob/main/examples/stm32f4/src/bin/multiprio.rs

logerup, 2024-02-06:
> this is the example you're looking for https://github.com/embassy-rs/embassy/blob/main/examples/stm32f4/src/bin/multiprio.rs

That's exactly it, thanks for the help. I was only looking in the H7 examples...

closed 2024-02-06 by logerup