embassy-rs/embassy #2538: embassy-net: allocate StackResources on the heap

dignifiedquire, 2024-02-06:
In order to be able to support many TCP sockets, I am trying to allocate `StackResources` on the heap, instead of statically. Unfortunately the `&'static mut StackResources`  argument to `Stack::new` makes it quite hard. Would it be possible to provide a constructor that owns the resources, similarly to how smoltcp uses `managed` to allow for that in a setup where `alloc` is possible?

dignifiedquire, 2024-02-06:
fyi for anyone trying to this as well, this works for now

```rust
let resources = make_static!(alloc::boxed::Box::new(
    StackResources::<WIFI_STACK_SIZE>::new()
));
let stack = &*make_static!(Stack::new(wifi_interface, config, resources, seed));
```

Dominaezzz, 2024-02-11:
Do you actually want https://doc.rust-lang.org/std/boxed/struct.Box.html#method.leak ?

Though it might be more interesting to have a `HeapResources`.

dignifiedquire, 2024-02-11:
using leak would be better, but it would be nice to be able to gurantee it is allocated on the heap in the first place, which Box:new in this form doesn’t (most compiler optimizations make it so in practice)

closed 2025-01-10 by Dirbaio