embassy-rs/embassy #2542: Unaligned buffer in embassy-boot async firmware updater for RP2040

chamburr, 2024-02-07:
When using the async firmware updater for RP2040, all operations to update the magic code fails with an unaligned error.

This can be traced to changes made in #2411, after which the aligned buffer must have a size of 4 due to READ_SIZE being the larger size, and being defined as 4 for RP2040.

https://github.com/embassy-rs/embassy/blob/2c5426aa5c41563f3b225d99d39793a765ec9204/embassy-boot/src/firmware_updater/asynch.rs#L229-L230

This becomes a problem for the set magic function, which calls the flash read function for both offset 0 (line 274) as well as 1 (line 278). This is because WRITE_SIZE is defined as 1 for RP2040.

https://github.com/embassy-rs/embassy/blob/2c5426aa5c41563f3b225d99d39793a765ec9204/embassy-boot/src/firmware_updater/asynch.rs#L273-L278

This eventually triggers an error in the flash background read function for RP2040, which requires that the offset is a multiple of 4 (lines 286 to 288).

https://github.com/embassy-rs/embassy/blob/2c5426aa5c41563f3b225d99d39793a765ec9204/embassy-rp/src/flash.rs#L270-L288

lulf, 2024-02-07:
Thanks for the bug report! I'm wondering if the best fix here is to make the magic of size `STATE::WRITE_SIZE.max(STATE::READ_SIZE))` all over then.

chamburr, 2024-02-07:
Hmm... This doesn't sound like a simple fix with READ_SIZE being different for sync and async flash. I suspect we also have to change the magic size in the blocking firmware updater and the boot loader, and it would be something like max of sync read, sync write, async read and async write sizes?

PR #2556 (2024-02-12): Fix unaligned buffer in async updater
This PR fixes #2542.

I didn't know how to implement the change in magic size everywhere as suggested by lulf in the issue. What I've done here is to keep the magic size at size of WRITE_SIZE, and slice the buffer for writes to also be size of WRITE_SIZE. To make sure that reads are aligned to READ_SIZE, I avoid the second read for progress whenever the data is available in the first read. I have tested this on RP2040 and it works.
closed 2024-02-12 by lulf