embassy-rs/embassy #2546: STM32H503: enabling RTC using LSI causes startup crashes and UB

tcbennun, 2024-02-08:
Been banging my head against a wall about this for a while now.

I have the NUCLEO-H503RB. If I take the `stm32h5` blinky example and make minimal changes:

- change the chip to the H503RB
- change task arena to 16384 or lower (to fit in the small memory of the H503)
- change the blinky pin to PA5 to match the board

... then I get a hard fault somewhere during RCC init, sometimes; other times, I get corruption on the stack.

By default, RTC is enabled by default and is set to source from the LSI. If I change the example main as follows:

```rust
    let mut config = Config::default();
    config.rcc.ls = LsConfig::off();
    let peris = embassy_stm32::init(config);
```

then everything appears to be okay again.

I see there's a notice in the code that there's a similar issue with L5. Wonder if it's the same thing.

tcbennun, 2024-02-16:
I've narrowed it down to these lines.

https://github.com/embassy-rs/embassy/blob/26fc17e8a7db0957896db28cb2299ab9cf89aea5/embassy-stm32/src/rcc/bd.rs#L206

Forcing a backup domain reset followed by immediate writes to RCC_BDCR is clearly problematic

Dirbaio, 2024-02-17:
I can't reproduce this on my NUCLEO-H503RB. Can you provide exact code (a branch on a fork for example) and exact `cargo run` command to reproduce? thanks!

tcbennun, 2024-02-21:
> I can't reproduce this on my NUCLEO-H503RB. Can you provide exact code (a branch on a fork for example) and exact `cargo run` command to reproduce? thanks!

Sorry, didn't see this until now, I'll provide something soon

I'm hoping I haven't damaged all my boards in some weird specific way...

tcbennun, 2024-02-22:
@Dirbaio 
https://github.com/tcbennun/nucleo-h503rb-minimal-rs

Dirbaio, 2024-02-22:
okay, I was able to reproduce the issue with your repo. thank you! :) 

Turns out the issue is this gem of an errata. If I set `memory.x` to only use SRAM1 the corruption no longer happens.

[image]

I'm ... not sure how to workaround that.

tcbennun, 2024-02-22:
Thank you so much for digging and finding that!

I'll need to work around it somehow, pretty mission critical. I reckon I can hack my use case so nothing critical is in SRAM2 at the point of backup domain reset (should only be during init). We shall see...

PR #2616 (2024-02-22): stm32/rcc: workaround nonsense RAM suicide errata on backup domain reset.
fixes  #2546
closed 2024-02-22 by Dirbaio