embassy-rs/embassy #2554: embassy-usb: Async control request handler

0e4ef622, 2024-02-10:
Currently, it does not seem possible to add custom `async fn` USB control request handlers. Is there any way to support this?

Dirbaio, 2024-02-10:
it's a bit complicated, control handlers use `dyn` and async traits can't use `dyn`. It'd require using actual generics, which i'm not sure how it'd work because there's N handlers, not just 1. Each class can add its own handler. And it'd probably make the type signature of `UsbDevice` much more complex.

what's your use case for async control handlers?

0e4ef622, 2024-02-10:
My control handler needs to talk to an spi device but at the same time, there's a loop future that runs every 5 ms which reads data from the same spi device.

I wanted to use an async mutex for this but that clearly won't work, so currently I'm just using a refcell and blocking calls in the handler. If it fails to borrow, I return `Rejected`.

Another solution I thought about was to have another loop and have the control handler communicate via channels to handle the request but that only works for control out not control in since I can't send return values.