embassy-rs/embassy #2557: Using the Ws2812 PIO example on a Ws2811

0xa10, 2024-02-10:
Hi - Im trying to get the [Ws2812 PIO example ](https://github.com/embassy-rs/embassy/blob/main/examples/rp/src/bin/pio_ws2812.rs)working using these leds:
https://www.pololu.com/product/2535

I tested one with the Adafruit_Neopixel arduino sketch running on a Pi Pico - the LED is powered from 5V and Im translating the 3.3V logic to 5V with a 74HCT245.
I'm able to control the RGB values of the LED and it behaves as expected.

However when using the PIO example the LED only switches between various intensities of red.

If instead of pushing u32 values into the FIFO I push u8 I am also able to control green and blue, but not red:
```
pub async fn write_direct(&mut self, words: &[u8]) {
        self.sm.tx().dma_push(self.dma.reborrow(), words).await;
    }
```

I think the issue is either in:
1. The Ws2811 timings are different - I tried extending the data bit cycle but it had no effect. Not sure why the NeoPixel library worked if this is the case though.
2. I am misunderstanding the FIFO behavior as its implemented in the example. Why does the word size matter here? Perhaps its affecting the framing somehow?

Any help would be greatly appreciated

CBJamo, 2024-02-10:
The first difference between the ws2812 and ws2811 is that the 12 expects the color in GRB order and the 11 expects RGB. That difference would mess with your colors, but doesn't fit the issue you're describing. To change this, r and g need to be swapped on line 104, like so: 
```rust
let word = (u32::from(colors[i].r) << 24) | (u32::from(colors[i].g) << 16) | (u32::from(colors[i].b) << 8);
```

Different bit timings might explain the issue you're seeing. There are several datasheets for the ws281x family, and the timings are all slightly different, except that the total bit time is the same. In the pio program we divide that time up into T1, T2, and T3, and they need to add up to 10 for the total time to be correct. The specific timings in the example come from the ws2812 example in the rp2040 datasheet.

I played around with them a bit on the ws2812b's I have and found that several timings worked. T1=3 T2=4, and T3=3 is the closest to the 12b's datasheet values. I suspect the neopixel lib uses different sub-bit timings and that's why it worked. Alternately, do you specify the led type you're using?

The fifo is 32 bits wide, but is configured to shift out 24 bits (8 bits each of r, g, and b) before advancing to the next word, this is set with the threshold field of the ShiftConfig struct. Also, it shifts out left, that's why the colors are shifted 24, 16, and 8 bits, rather than 16, 8, and 0.

The ws281x series leds don't have any framing, other than the reset time being used to update the colors. You can think of the led string as a shift register, where you just shift the entire color string out then (with the reset time) update the entire string. If you shift out the wrong number of bytes, your colors will be interpreted wrong. I think if an individual LED gets less than 24 bits it treats the colors it didn't get as 0, but I'm not sure about that.

0xa10, 2024-02-10:
Thanks for the help.
These are the leds Im using:
https://www.pololu.com/product/2535

One datasheet for the WS2811 I took a look at does specify slightly different timings, but the product page shows the same ~1.2uS frame duration as you'd expect.

I took a look at the NeoPixel source code and it does indeed seem to use the same timings:
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/rp2040_pio.h

Other than that, the only differences I could find is that the NeoPixel code sets the ShiftConfig threshold to 8 bits instead of 24, and uses 3 consecutive blocking writes instead of DMA - but I tried that on my LED and it has the same effect of just setting the red intensity.

What's really confusing me is that `StateMachineTx<'d, PIO, SM>::push` takes a u32 always but `StateMachineTx<'d, PIO, SM>::dma_push` is generic over the word size, and if I call `dma_push` with an array of u8's I can get only blue and green to light up, where as if I send it as a single u32 only red lights up at various intensities.

Just to illustrate - this is how I modified the write function (I realize this doesn't make sense for more than one led but thats what Im testing now):
```rust
    pub async fn write(&mut self, colors: &[RGB8; N]) {
        // Precompute the word bytes from the colors
        for i in 0..N {
            let words = [colors[i].g, colors[i].r, colors[i].b];
            // DMA transfer
            self.sm.tx().dma_push(self.dma.reborrow(), &words).await;
        }
    }
}
```

I also noticed just now that changing the ShiftConfig threshold causes the behavior to change - when changing the threshold to 8 I get different color combos - green and yellow, so this is definitely something to do with framing and timing, perhaps some issue with the way the FIFO is packed?

Should lowering the auto-pull threshold to 8 even make a difference? Doesn't it just mean the FIFO will be pulled 3 times instead of once for each LED? Or does the auto-pull cause a whole word to be read regardless of the actual threshold?

CBJamo, 2024-02-10:
> `StateMachineTx<'d, PIO, SM>::push` takes a u32 

The fifo is 32 bits wide. If you want to squish smaller types into one word you have to do that in the CPU.

> `StateMachineTx<'d, PIO, SM>::dma_push` is generic over the word size

I didn't write this, but I'm reasonably certain this is simply for convenience. No matter what type you send, each element takes it's own fifo slot. I suspect that the other bits are zeroed, but they might have random data.

> Should lowering the auto-pull threshold to 8 even make a difference? Doesn't it just mean the FIFO will be pulled 3 times instead of once for each LED? Or does the auto-pull cause a whole word to be read regardless of the actual threshold?

The latter, the OUT instruction pulls some number of bits (1 in our case) out of the output shift register (OSR) into some destination (the X register in our case). The PULL instruction moves a word from the FIFO to the OSR. Auto pull does the same as a PULL instruction every `threshold` bits.

The ws2812 pio program simply takes the bits from the FIFO and serializes it according to the ws281x timings. This example, and the original example from the 2040 ds, pack the color data for an led into a word, using a single FIFO slot per led. The adafruit code uses a FIFO slot per byte of the rgb data, using 3 FIFO slots per led. 

If your data is already organized in a big 1d array of `[r1,g1,b1,r2,g2,b2,...]` then it would be very natural to use DMA and the byte method, if your data is in some kind of rgb struct then it's a bit nicer to use the 24 bit method. Whichever you use, the `ShiftConfig.threshold` value will need to match the data you're putting in the FIFO.

The pololu documentation says these leds are configured to use the 800kHz/fast mode, but maybe they're actually 400kHz? Try changing `ws2812_freq` on line 79 to 400 and see if that fixes it. The way that you're getting some colors has me wondering if they're getting garbled data.

0xa10, 2024-02-11:
So after reading into the DMA code and datasheet - it seems that when DMA pushing partial words to the FIFO, the content will get replicated to fill the FIFO slot, which explains why the behavior is different, but I have no idea why in the full word case I'm only getting red.

The thing I still don't fully understand is the autopull behavior - if my threshold is 24 bits, that means that after 24 `out x, 1` instructions the OSR will be filled with 32 bits from the next FIFO slot? What happens to the "remaining" 8 bits in the OSR? Are they overwritten?

CBJamo, 2024-02-11:
I didn't know the DMA would replicate the data, that's certainly more useful than zeros or random data.

Yes, any time a PULL instruction (or an auto pull happens) a full word is written from the FIFO to the OSR.

0xa10, 2024-02-11:
Well, I finally gave in and hooked up a scope and saw that the timings were very slightly off.
I reduced the 1-bit duration by one cycle and the LED started behaving correctly, even though the timings were still not quite to spec.

Inexplicably - after having reverted to the original code, with the original timings - the LED now works perfectly fine!
I genuinely have no clue what could have caused the original issue, my only guess is breadboard voodoo 🤷

Sorry for the unscientific conclusion

closed 2024-02-11 by 0xa10