embassy-rs/embassy #2566: nrf52840 usb: index out of bounds: the len is 8 but the index is 8

vDorst, 2024-02-13:
I am trying to create a usb soundcard, first I started with a blackpill stm32f401.
But I had issues with sample delivery it was every 2mS instead of the wanted 1mS.
So I thought to rule out the stm32 usb stack I port the basic code over to a nrf52820.

But I hit a `index out of bounds: the len is 8 but the index is 8` error.

Code: https://github.com/vDorst/stm32_audio/tree/nrf 
own embassy brach but it is not needed I think: https://github.com/vDorst/embassy/tree/i2s-no-mclk

For the UAC1.0 I only using a usb controll and isoch out endpoint.

Cargo run log
```
warning: `stm32_audio` (bin "stm32_audio") generated 12 warnings (run `cargo fix --bin "stm32_audio"` to apply 1 suggestion)
    Finished release [optimized + debuginfo] target(s) in 14.77s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/release/stm32_audio`
      Erasing ✔ [00:00:01] [################################################################################################################################] 36.00 KiB/36.00 KiB @ 26.70 KiB/s (eta 0s )
  Programming ✔ [00:00:02] [################################################################################################################################] 36.00 KiB/36.00 KiB @ 16.63 KiB/s (eta 0s )    Finished in 3.572s
0.000000 INFO  Enabling ext hfosc...
└─ stm32_audio::____embassy_main_task::{async_fn#0} @ src/main.rs:176 
0.000274 INFO  30 == 30
└─ stm32_audio::usb_audio_class::{impl#0}::new @ src/usb_audio_class.rs:214 
0.000366 INFO  USB: device_descriptor used: 18
└─ embassy_usb::builder::{impl#1}::build @ /home/r/dev/rust/embassy/embassy-usb/src/fmt.rs:143 
0.000427 INFO  USB: config_descriptor used: 100 [09, 02, 64, 00, 02, 01, 00, 80, 32, 09, 04, 00, 00, 00, 01, 01, 00, 00, 09, 24, 01, 00, 01, 1e, 00, 01, 01, 0c, 24, 02, 01, 01, 01, 03, 02, 03, 00, 00, 00, 09, 24, 03, 03, 02, 03, 01, 01, 00, 09, 04, 01, 00, 00, 01, 02, 00, 00, 09, 04, 01, 01, 01, 01, 02, 00, 00, 07, 24, 01, 01, 01, 01, 00, 0b, 24, 02, 01, 02, 02, 10, 01, 80, bb, 00, 09, 05, 08, 09, c8, 00, 01, 00, 00, 07, 25, 01, 01, 02, 04, 00]
└─ embassy_usb::builder::{impl#1}::build @ /home/r/dev/rust/embassy/embassy-usb/src/fmt.rs:143 
0.000915 INFO  USB: bos_descriptor used: 12
└─ embassy_usb::builder::{impl#1}::build @ /home/r/dev/rust/embassy/embassy-usb/src/fmt.rs:143 
0.000976 INFO  USB: msos_descriptor used: 0
└─ embassy_usb::builder::{impl#1}::build @ /home/r/dev/rust/embassy/embassy-usb/src/fmt.rs:143 
0.001007 INFO  USB: control_buf size: 256
└─ embassy_usb::builder::{impl#1}::build @ /home/r/dev/rust/embassy/embassy-usb/src/fmt.rs:143 
0.001098 INFO  write done
└─ stm32_audio::____embassy_main_task::{async_fn#0} @ src/main.rs:227 
0.001129 INFO  init usb runner
└─ stm32_audio::__usb_runner_task::{async_fn#0} @ src/main.rs:82  
0.002197 INFO  EN: true
└─ stm32_audio::usb_audio_class::{impl#8}::enabled @ src/usb_audio_class.rs:653 
loop 00
0.011169 INFO  Wait for USB Audio samples
└─ stm32_audio::__usb_samples_task_task::{async_fn#0} @ src/main.rs:123 
0.316497 INFO  AUC: Reset
└─ stm32_audio::usb_audio_class::{impl#8}::reset @ src/usb_audio_class.rs:477 
0.316528 INFO  ALT: inf# 0 a_setting: 0
└─ stm32_audio::usb_audio_class::{impl#8}::set_alternate_setting @ src/usb_audio_class.rs:661 
0.316589 INFO  ALT: inf# 1 a_setting: 0
└─ stm32_audio::usb_audio_class::{impl#8}::set_alternate_setting @ src/usb_audio_class.rs:661 
0.508300 INFO  AUC: Reset
└─ stm32_audio::usb_audio_class::{impl#8}::reset @ src/usb_audio_class.rs:477 
0.508331 INFO  ALT: inf# 0 a_setting: 0
└─ stm32_audio::usb_audio_class::{impl#8}::set_alternate_setting @ src/usb_audio_class.rs:661 
0.508361 INFO  ALT: inf# 1 a_setting: 0
└─ stm32_audio::usb_audio_class::{impl#8}::set_alternate_setting @ src/usb_audio_class.rs:661 
0.734039 INFO  CFG: true
└─ stm32_audio::usb_audio_class::{impl#8}::configured @ src/usb_audio_class.rs:657 
0.742034 INFO  ALT: inf# 1 a_setting: 0
└─ stm32_audio::usb_audio_class::{impl#8}::set_alternate_setting @ src/usb_audio_class.rs:661 
0.759063 ERROR panicked at /home/r/dev/rust/embassy/embassy-nrf/src/usb/mod.rs:359:21:
index out of bounds: the len is 8 but the index is 8
└─ panic_probe::print_defmt::print @ /home/r/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:104 
```

dmesg log
```
[  243.878189] usb 3-2.1: new full-speed USB device number 8 using xhci_hcd
[  244.105111] usb 3-2.1: New USB device found, idVendor=c0de, idProduct=cafe, bcdDevice= 0.10
[  244.105121] usb 3-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  244.105124] usb 3-2.1: Product: USB-Audio
[  244.105127] usb 3-2.1: Manufacturer: Embassy
[  244.105130] usb 3-2.1: SerialNumber: 12345678
[  254.251644] usb 3-2.1: 1:1: cannot set freq 48000 to ep 0x8
[  264.515289] usb 3-2.1: 1:1: usb_set_interface failed (-110)
[  269.635191] usb 3-2.1: 1:1: usb_set_interface failed (-110)
[  274.755104] usb 3-2.1: 1:1: usb_set_interface failed (-110)
[  279.875074] usb 3-2.1: 1:1: usb_set_interface failed (-110)
```

Dirbaio, 2024-02-13:
can you post trace-level logs?

vDorst, 2024-02-13:
I found the issue but I think it is not quick to solve.
This comment in the code gives me the clue. [iso](https://github.com/embassy-rs/embassy/blob/main/embassy-nrf/src/usb/mod.rs#L769)
Isochronous have there own hardware registers.
It seems that is currently not supported in the nrf hal.