embassy-rs/embassy #2640: how to get timer tick value in stm32 embassy?

learncodingforweb, 2024-02-29:
I am running embassy task. how i get value of timer tick value inside task. so that if timer tick expire after some n ms. then task will perform some action after n ms. i am not able to find any solution? How can i do this in embassy.? 
```
#[embassy_executor::task]
async fn rf_recieve(mut rx: UartRx<'static, USART3, DMA1_CH1>) {
    let mut buf: [u8; config::RF_BUFF_SIZE];
    let mut rbuf: [u8; 1] = [0; 1];
    let mut ind = 0;
    defmt::info!("single mode selected");
    buf = [0; config::RF_BUFF_SIZE];

    loop {
        let r = rx.read(&mut rbuf).await;
        if r.is_ok() {
            buf[ind] = rbuf[0];
            if ind < config::RF_BUFF_SIZE {
                ind += 1;
                defmt::info!("serial data {} => {}", ind, rbuf);
            }
            if ind == config::RF_BUFF_SIZE {
                let fd = bsp::rf_parse::rf_filter(&buf);
                defmt::info!("{:?}", fd);                
            }
        } else {
            defmt::info!("serial data read error");
        }
    }
}
```

vDorst, 2024-03-01:
The main [page has](https://embassy.dev/) has an example that used `Timer`.
Other option is `Ticker`. See also the documentation https://docs.embassy.dev/embassy-time/git/default/struct.Ticker.html

```
#[embassy_executor::task]
async fn blink(pin: AnyPin) {
    let mut led = Output::new(pin, Level::Low, OutputDrive::Standard);

    loop {
        // Timekeeping is globally available, no need to mess with hardware timers.
        led.set_high();
        Timer::after_millis(150).await;
        led.set_low();
        Timer::after_millis(150).await;
    }
}```

lulf, 2024-03-02:
@learncodingforweb If you need the tick value, you can use `Instant` (https://docs.rs/embassy-time/0.3.0/embassy_time/struct.Instant.html#method.as_ticks): `Instant::now().as_ticks()`

closed 2024-03-02 by lulf