embassy-rs/embassy #2674: could not find `executor` in `embassy`

0xkelvin, 2024-03-09:
hello, 

I am trying to do basic "hello world" with embassy on esp32c6, but i got error "could not find `executor` in `embassy`"

my cargo.toml : 
```
[package]
name = "c6-hello"
version = "0.1.0"
authors = ["Kelvin <quocvm1@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.48", default-features = false }
esp-backtrace       = { version = "0.11.0", features = ["esp32c6", "exception-handler", "panic-handler", "println"] }
embassy-executor    = { version = "0.5.0", features = ["nightly", "integrated-timers", "arch-riscv32", "executor-thread"] }
embassy-sync        = "0.5.0"
embassy-time        = "0.3.0"
embassy-time-driver = { version = "0.1.0", optional = true }
embedded-graphics   = "0.8.1"
embedded-hal        = "1.0.0"
embedded-hal-02     = { version = "0.2.7", package = "embedded-hal" }
embedded-hal-async  = "1.0.0"
embedded-hal-bus    = "0.1.0"
embedded-io-async   = "0.6.1"
esp32c6-hal = { version = "0.8.0", features = ["async", "embassy", "embassy-time-timg0"] }
esp-println = { version = "0.9.0", features = ["esp32c6"] }

[build-dependencies]
embuild = "0.31.3"

```

great thanks for any comments

DeppLearning, 2024-03-13:
Not an expert, but I guess you might be missing a feature flag, [see](https://github.com/esp-rs/esp-hal/blob/6046528e0b6ab80402d3c69b9e53ff050c5f8a45/esp-hal/Cargo.toml#L183)

manuelbaez, 2024-03-16:
Hey, not sure if this helps but you could also use the esp-hal crate for no_std then need to enable these features and should work:
 `features = [
    "esp32c6",
    "embassy",
    "embassy-time-timg0",
    "embassy-executor-thread",
    "embassy-executor-interrupt"
]`
I think you could also remove the esp32c6-hal crate since it's deprecated for esp-hal.

0xkelvin, 2024-03-18:
thanks, that's works

closed 2024-03-18 by 0xkelvin