embassy-rs/embassy #2679: embassy::futures::block_on does not work

AliMoal, 2024-03-10:
I'm using nRF5232 and I need to use `block_on` function to use an async function in a sync one.
But the program stucks in this function and never returns. I checked my function with `.await` and it works without any delays but with `block_on` function it stucks. I cannot use future crate because it needs std feature.

Dirbaio, 2024-03-10:
`block_on` does work but it blocks all tasks from running, so you can have deadlocks if the thing you're `block_on`ing requires some background task for example to handle events. This can happen for example with `nrf-softdevice` Flash stuff.

AliMoal, 2024-03-10:
> `block_on` does work but it blocks all tasks from running, so you can have deadlocks if the thing you're `block_on`ing requires some background task for example to handle events. This can happen for example with `nrf-softdevice` Flash stuff.

Thanks for your quick reply.
As you said I'm working with `nrf-softdevice` Flash stuff. So, what should I do in this situation?

lulf, 2024-03-12:
If you need to use the flash in a blocking way, you need to run the softdevice a bit differently. You can do that by running these two methods in separate embassy executors. A high priority executor running the run_soc, and a low priority executor running the run_ble task: https://github.com/embassy-rs/nrf-softdevice/blob/master/nrf-softdevice/src/softdevice.rs#L327-L335 . This will allow the flash operation events to be processed by the higher priority executor, assuming you're using the flash inside a lower priority task.

lulf, 2024-03-21:
Closing this as I think you got it working. Feel free to reopen if not!

closed 2024-03-21 by lulf

Leandros, 2024-12-02:
@lulf:

> A high priority executor running the run_soc, and a low priority executor running the run_ble task

How would I do that with `SoftDevice` being `!Send` and `!Sync`? 

Something like this, from the embassy examples, doesn't work:

```rust
#[entry]
fn main() -> ! {
    // ...
    let sd = SoftDevice::enable(&confgi);

    interrupt::SWI1_EGU1.set_priority(Priority::P6);
    let spawner = EXECUTOR_HIGH.start(interrupt::SWI1_EGU1);
    spawner.must_spawn(sd_task_high(sd)); // <-- `*mut ()` cannot be shared between threads safely
}
```

And if I create the `SoftDevice` in the executor, I can't get it out to run the low-priority executor. Am I missing something here?