embassy-rs/embassy #2681: examples/rp/wifi* issue with mismatching crate/code

RafaelKoch, 2024-03-10:
When trying to make this work in a new project: https://github.com/embassy-rs/embassy/blob/main/examples/rp/src/bin/wifi_blinky.rs

I get this:

```rust
 Compiling alarmclock v0.1.0 (C:\git\pico-rust-fiddling\alarmclock)
error[E0107]: struct takes 1 generic argument but 0 generic arguments were supplied
   --> src\bin\wifi_blinky.rs:24:51
    |
24  | async fn wifi_task(runner: cyw43::Runner<'static, Output<'static>, PioSpi<'static, PIO0, 0, DMA_CH0>>) -> ! {
    |                                                   ^^^^^^ expected 1 generic argument
    |
note: struct defined here, with 1 generic parameter: `T`
   --> C:\Users\[]\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embassy-rp-0.1.0\src\gpio.rs:347:12
    |
347 | pub struct Output<'d, T: Pin> {
    |            ^^^^^^     -
help: add missing generic argument
    |
24  | async fn wifi_task(runner: cyw43::Runner<'static, Output<'static, T>, PioSpi<'static, PIO0, 0, DMA_CH0>>) -> ! {
    |                                                                 +++

error[E0107]: struct takes 4 generic arguments but 3 generic arguments were supplied
  --> src\bin\wifi_blinky.rs:24:68
   |
24 | async fn wifi_task(runner: cyw43::Runner<'static, Output<'static>, PioSpi<'static, PIO0, 0, DMA_CH0>>) -> ! {
   |                                                                    ^^^^^^          ----  -  ------- supplied 3 generic arguments
   |                                                                    |
   |                                                                    expected 4 generic arguments
   |
note: struct defined here, with 4 generic parameters: `CS`, `PIO`, `SM`, `DMA`
  --> C:\Users\[]\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cyw43-pio-0.1.0\src\lib.rs:17:12
   |
17 | pub struct PioSpi<'d, CS: Pin, PIO: Instance, const SM: usize, DMA> {
   |            ^^^^^^     --       ---            ---------------  ---
help: add missing generic argument
   |
24 | async fn wifi_task(runner: cyw43::Runner<'static, Output<'static>, PioSpi<'static, PIO0, 0, DMA_CH0, DMA>>) -> ! {
   |                                                                                                    +++++
```

Whereas when cloning embassy repo and running the very same it works. I believe this is down to the code in embassy repo and the crates not matching up. These two here have at least one struct different when using crates instead of the code directly:

```toml
embassy-rp = { version = "0.1.0", path = "../../embassy-rp", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl"] }
cyw43-pio = { version = "0.1.0", path = "../../cyw43-pio", features = ["defmt", "overclock"] }
```

Same goes for the other wifi-samples, they all do not compile when using a create.

Dirbaio, 2024-03-10:
Examples in the repo work with the version of `embassy-rp` in the repo, which has changes that are not present in the `0.1.0` version released on crates.io.

If you want to get examples that work with the crates.io versions, you have to go back in the repo history to when embassy-rp 0.1.0 was released. there's an `embassy-rp-v0.1.0` tag for that: https://github.com/embassy-rs/embassy/tree/embassy-rp-v0.1.0 . Then, copy the example from there.

Alternatively, you can use `[patch.crates-io]` in `Cargo.toml` to replace all embassy crates with references to the git repo.

1-rafael-1, 2024-03-11:
Ah, that makes sense - Thanks! Will try to reference the repo then, I guess.
I will close the issue.

closed 2024-03-11 by RafaelKoch