embassy-rs/embassy #2684: STM32WB5x wpan examples don't seem to work

DeppLearning, 2024-03-11:
I'm trying to run the [stm32wb mac_ffd example](https://github.com/embassy-rs/embassy/blob/main/examples/stm32wb/src/bin/mac_ffd_net.rs) on my NUCLEO-WB55RG board but get stuck [here](https://github.com/embassy-rs/embassy/blob/ac06ca2fa0aa061a0053336433fa9cbfad5212ca/examples/stm32wb/src/bin/mac_ffd_net.rs#L73). I followed the instructions in the comments and was able to "Start Wireless Stack" eventually in Stm32 Cube Programmer. I am missing the [teleprobe](https://github.com/embassy-rs/embassy/blob/ac06ca2fa0aa061a0053336433fa9cbfad5212ca/examples/stm32wb/.cargo/config.toml#L4C12-L4C21) tool, so I used the commented line above, maybe that is an issue?

This is the log output
```
0.000000 TRACE BDCR ok: 00008113
└─ embassy_stm32::rcc::bd::{impl#3}::init @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.000000 DEBUG rcc: Clocks { hclk1: Some(Hertz(64000000)), hclk3: Some(Hertz(64000000)), hse: Some(Hertz(32000000)), hsi: None, hsi48: Some(Hertz(48000000)), lse: None, lsi: None, msi: None, pclk1: Some(Hertz(64000000)), pclk1_tim: Some(Hertz(64000000)), pclk2: Some(Hertz(64000000)), pclk2_tim: Some(Hertz(64000000)), pll1_p: Some(Hertz(64000000)), pll1_q: Some(Hertz(48000000)), pllsai1_p: None, pllsai1_q: None, pllsai1_r: None, rtc: Some(Hertz(32768)), sai1_extclk: None, sys: Some(Hertz(64000000)) }
└─ embassy_stm32::rcc::set_freqs @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:130
0.000000 INFO  Hello World!
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:55
0.000122 TRACE ipcc: ch 1: wait for rx occupied
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.003387 TRACE ipcc: ch 1: read data
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.003417 INFO  sys event: [0, 146, 0]
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:63
0.003479 TRACE ipcc: ch 1: send data
└─ embassy_stm32::ipcc::{impl#2}::send::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.003540 TRACE ipcc: ch 1: wait for tx free
└─ embassy_stm32::ipcc::{impl#2}::flush::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.003601 TRACE ipcc: ch 3: send data
└─ embassy_stm32::ipcc::{impl#2}::send::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
```

This is my memory.x file:
```
/* Change this as required for your MCU */

MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 256K 
  RAM : ORIGIN = 0x20000000, LENGTH = 64K
}
```
I guess instead of LENGTH = 256K, I should use 226K but changing that didn't get me any other results.

I get similar results from running the other examples. Of the radio ones, only the `tl_mbox` example finishes successfully. I also tried following the flashing procedure described in the release-notes file mentioned in the comments (which is slightly different) of the embassy examples, but got the same results. Any ideas what I'm doing wrong? I am new to IEEE 802.15.4 and STM32 so I might have missed sth very obvious.

DeppLearning, 2024-03-13:
I tried again, the above result was probably due to some problem with the binary blobs which got fixed after trying the main branch and before trying 8b36a32ed5d834b23e970d5b723dd7df1f1c94a2. I can reproduce this, i.e. by "Delete Firmaware" in CubeProgrammer.

However, now after I got the flashing of binary blobs fixed, when I start the mac_ffd example on one board and the mac_rfd example on another, I'll get a panic on the board running mac_ffd as soon as it receives the message from the board running mac_rfd:

```
0.000000 TRACE BDCR ok: 00008113
└─ embassy_stm32::rcc::bd::{impl#3}::init @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.000000 DEBUG rcc: Clocks { hclk1: Some(Hertz(64000000)), hclk3: Some(Hertz(64000000)), hse: Some(Hertz(32000000)), hsi: None, hsi48: Some(Hertz(48000000)), lse: None, lsi: None, msi: None, pclk1: Some(Hertz(64000000)), pclk1_tim: Some(Hertz(64000000)), pclk2: Some(Hertz(64000000)), pclk2_tim: Some(Hertz(64000000)), pll1_p: Some(Hertz(64000000)), pll1_q: Some(Hertz(48000000)), pllsai1_p: None, pllsai1_q: None, pllsai1_r: None, rtc: Some(Hertz(32768)), sai1_extclk: None, sys: Some(Hertz(64000000)) }
└─ embassy_stm32::rcc::set_freqs @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:130
0.000000 INFO  Hello World!
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:67
0.000122 TRACE ipcc: ch 1: wait for rx occupied
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.003387 TRACE ipcc: ch 1: read data
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.003417 INFO  sys event: [0, 146, 0]
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:75
0.003479 TRACE ipcc: ch 1: send data
└─ embassy_stm32::ipcc::{impl#2}::send::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.003509 TRACE ipcc: ch 1: wait for tx free
└─ embassy_stm32::ipcc::{impl#2}::flush::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.003601 TRACE ipcc: ch 3: send data
└─ embassy_stm32::ipcc::{impl#2}::send::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.021972 INFO  initialized mac: Ok(ShciSuccess)
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:80
0.022033 INFO  resetting
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:82
0.022064 TRACE ipcc: ch 2: send data
└─ embassy_stm32::ipcc::{impl#2}::send::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.022094 TRACE ipcc: ch 2: wait for tx free
└─ embassy_stm32::ipcc::{impl#2}::flush::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.031341 TRACE ipcc: ch 2: wait for rx occupied
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.031402 TRACE ipcc: ch 2: read data
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.031433 INFO  MlmeResetCnf(ResetConfirm { status: Success, a_stuffing: [0, 0, 0] })
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:90
0.031494 TRACE mac drop event
└─ embassy_stm32_wpan::sub::mac::{impl#1}::drop_event_packet @ /home/me/git/embassy/embassy-stm32-wpan/src/fmt.rs:117
0.031524 TRACE ipcc: ch 2: read data
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.031555 TRACE ipcc: ch 2: clear rx
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.031585 TRACE ipcc: ch 2: wait for rx occupied
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.031646 INFO  setting extended address
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:92
0.031677 TRACE ipcc: ch 2: send data
└─ embassy_stm32::ipcc::{impl#2}::send::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.031707 TRACE ipcc: ch 2: wait for tx free
└─ embassy_stm32::ipcc::{impl#2}::flush::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.032806 TRACE ipcc: ch 2: read data
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.032836 INFO  MlmeSetCnf(SetConfirm { status: Success, pin_attribute: ExtendedAddress, a_stuffing: [0, 0] })
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:101
0.032897 TRACE mac drop event
└─ embassy_stm32_wpan::sub::mac::{impl#1}::drop_event_packet @ /home/me/git/embassy/embassy-stm32-wpan/src/fmt.rs:117
0.032958 TRACE ipcc: ch 2: read data
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.032989 TRACE ipcc: ch 2: clear rx
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.033020 TRACE ipcc: ch 2: wait for rx occupied
└─ embassy_stm32::ipcc::{impl#2}::receive::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.033050 INFO  setting short address
└─ mac_ffd::____embassy_main_task::{async_fn#0} @ src/bin/mac_ffd.rs:103
0.033111 TRACE ipcc: ch 2: send data
└─ embassy_stm32::ipcc::{impl#2}::send::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
0.033142 TRACE ipcc: ch 2: wait for tx free
└─ embassy_stm32::ipcc::{impl#2}::flush::{async_fn#0} @ /home/me/git/embassy/embassy-stm32/src/fmt.rs:117
[... cut ...]
        at /home/me/git/embassy/embassy-executor/src/raw/mod.rs:526:9
  12: embassy_executor::arch::thread::Executor::run
        at /home/me/git/embassy/embassy-executor/src/arch/cortex_m.rs:106:21
  13: mac_ffd::__cortex_m_rt_main
        at src/bin/mac_ffd.rs:31:1
  14: main
        at src/bin/mac_ffd.rs:31:1
  15: Reset
(HOST) ERROR the program panicked
```

The same thing doesn't happen on 8b36a32ed5d834b23e970d5b723dd7df1f1c94a2 where I'll get the `Hello from embassy` message on the receiver and both programs keep running without panic.

DeppLearning, 2024-06-26:
I'm still seeing this on latest master. If I delete this line, the problem is gone: https://github.com/embassy-rs/embassy/blob/a2acb3e3dceddb4752f8fb1c17aa65e1959a2180/examples/stm32wb/src/bin/mac_ffd.rs#L153. 

I think it's just a problem with logging the `MacEvent::McpsDataInd` variant. Maybe there are others, didn't try all of them but this one seems to have a problem.

vvjcarter, 2025-01-24:
Can anyone shed a light on this? I am trying to understand why formating DataIndication leads to a hardfault but I was not very successfull with that.

Legorooj, 2025-03-31:
I have a similar issue, but instead of an error it was just hanging for me. I'm not running the example though, I'm running a modified version of it which I can try to get a minimal reproduction of if needed.