embassy-rs/embassy #2687: Example `eth_client.rs` not building for target stm32h730vb`

pkoevesdi, 2024-03-11:
I can build fine the example with the default target:
```
git clone https://github.com/embassy-rs/embassy
cd embassy/examples/stm32h7
cargo build --bin eth_client
```
but as soon as I change the target in the feature of `embassy-stm32` in `Cargo.toml` to `stm32h730vb`, I get the following error chain:
```
   Compiling embassy-stm32 v0.1.0 (/home/gauner/Downloads/embassy/embassy-stm32)
error[E0412]: cannot find type `RNG` in module `$crate::interrupt::typelevel`
  --> src/bin/eth_client.rs:22:5
   |
22 |     RNG => rng::InterruptHandler<peripherals::RNG>;
   |     ^^^ not found in `$crate::interrupt::typelevel`
   |
help: consider importing one of these items
   |
4  + use crate::peripherals::RNG;
   |
4  + use embassy_stm32::peripherals::RNG;
   |

error[E0277]: the trait bound `Irqs: Binding<embassy_stm32::interrupt::typelevel::HASH_RNG, embassy_stm32::rng::InterruptHandler<RNG>>` is not satisfied
  --> src/bin/eth_client.rs:60:35
   |
60 |     let mut rng = Rng::new(p.RNG, Irqs);
   |                   --------        ^^^^ the trait `Binding<embassy_stm32::interrupt::typelevel::HASH_RNG, embassy_stm32::rng::InterruptHandler<RNG>>` is not implemented for `Irqs`
   |                   |
   |                   required by a bound introduced by this call
   |
   = help: the following other types implement trait `Binding<I, H>`:
             <Irqs as Binding<embassy_stm32::interrupt::typelevel::ETH, embassy_stm32::eth::InterruptHandler>>
             <Irqs as Binding<{type error}, embassy_stm32::rng::InterruptHandler<RNG>>>
note: required by a bound in `embassy_stm32::rng::Rng::<'d, T>::new`
  --> /home/gauner/Downloads/embassy/embassy-stm32/src/rng.rs:53:20
   |
51 |     pub fn new(
   |            --- required by a bound in this associated function
52 |         inner: impl Peripheral<P = T> + 'd,
53 |         _irq: impl interrupt::typelevel::Binding<T::Interrupt, InterruptHandler<T>> + 'd,
   |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `Rng::<'d, T>::new`

Some errors have detailed explanations: E0277, E0412.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `embassy-stm32h7-examples` (bin "eth_client") due to 2 previous errors
```
When I change line 22 in `src/bin/eth_client.rs` 
```
-    RNG => rng::InterruptHandler<peripherals::RNG>;
+    HASH_RNG => rng::InterruptHandler<peripherals::RNG>;
```
and correct some pin assignments, it seemingly compiles.
(I get a different error message from the linker rust-lld then: "[stuff] will not fit in region 'FLASH'" - but that might be another story.)

Dirbaio, 2024-03-11:
This is expected, different chips have different interrupt names. If you change the chip it's possible you'll have to adjust the example code a bit.

I don't think this is something we can "fix" in the examples, other than maintaining examples for all 1400+ STM32 chips, which is not feasible.

pkoevesdi, 2024-03-11:
Ok, I understand, thanks for the clarification. 
It took me hours to figure this out, had to dig deep inside the `stm32-metapac` - because I (still) don't understand, what I'm doing. ;-) Maybe consider some generic hint/tutorial for embassy beginners like me how to use the examples with other chips, and what to look for to adjust?

closed 2024-03-11 by pkoevesdi