embassy-rs/embassy #2692: latest embassy: USB: ERROR received SETUP before previous finished processing

vDorst, 2024-03-12:
I am working on a STM32F401 with I2S audio dac to create a USB-audio jack.
With some modification on the embassy side, like changing i2s driver so it uses a ringbuffer and run without mclk, usb side better implement isochronous support and extra parameters needed for the USB audio class.
USB side, it used isochronous OUT for the audio samples and isochronous IN for the feedback real sample-rate.

This all is working on the "stable" embassy.
But I want it in the head version of embassy, because the DMA has changed and also to make a PR.

"stable":
* project: https://github.com/vDorst/stm32_audio/commits/current
* embassy: https://github.com/vDorst/embassy/tree/i2s-no-mclk

"latest":
* project: https://github.com/vDorst/stm32_audio/commits/main/
* embassy: https://github.com/vDorst/embassy/tree/latest

After the conversion I am stuck at usb "ERROR received SETUP before previous finished processing".
I can reproduce it every time and happens with stable 1.76 and nightly(10-03-2024) rust.
PC: linux 6.7.7 kernel.

Any idea why this is happening?

Console trace latest: [log-head.log](https://github.com/embassy-rs/embassy/files/14578216/log-head.log)
Console trace stable: [log-stable.log](https://github.com/embassy-rs/embassy/files/14578217/log-stable.log)

Dirbaio, 2024-03-12:
the issue is: when a SETUP packet is received, the irq stores the packet in a buffer and wakes the USB task  https://github.com/embassy-rs/embassy/blob/95234cddbac6f21fce0f5df510d49816f343b87d/embassy-stm32/src/usb_otg/usb.rs#L70
then the USB task needs to actually process it https://github.com/embassy-rs/embassy/blob/95234cddbac6f21fce0f5df510d49816f343b87d/embassy-stm32/src/usb_otg/usb.rs#L1319

The error happens when we receive a 2nd SETUP packet while the previous SETUP packet is still in the buffer, meaning that for some reason the USB task didn't get to process it.

are you sure you have no interrupt stuck preventing the USB task from running?

for example this looks suspicious https://github.com/vDorst/stm32_audio/blob/main/src/timer2.rs#L137

Dirbaio, 2024-03-12:
btw will you PR usb audio support? :eyes: 

it'd be aweoms to have it upstream with audio class implemented in embassy-usb

vDorst, 2024-03-12:
I am not sure.
But moving the TMR2 interrupt enable after the usb spawn task and add extra delay of 10 sec, it works again.
So that just seems confirming your suspicion.
You're definitely pointing me in the direction!

I would like to make a PR to add basic support for UAC.

closed 2024-03-12 by vDorst