embassy-rs/embassy #2696: UID: H5 doesn't allow 8bit reads from OTP

Dirbaio, 2024-03-14:
from the RM:

[image]

so this is probably wrong, because there's no guarantee user code will get optimized to doing wider reads. I think we should change the API to return a copy instead.

https://github.com/embassy-rs/embassy/blob/bbcab556c8a4514cd9ceda49d64c5644c82ba9e8/embassy-stm32/src/uid.rs#L5

tcbennun, 2024-03-14:
can confirm 8-bit reads on the ref returned by `uid()` is a crash on H503

eZioPan, 2024-03-17:
And the same thing happens to H562 too.

I also find metapac generate a [UID peripheral for f030rc]. But accroding to RM0360, there is no UID peripheral, and `0x1fff_f7ac` seems point to `System memory`?

[image]

I haven't check other MCUs, don't know if this happens to others.

---

For MCU that has a "Unique device ID" peripheral, should we generate fields and registers for it, not just a simple pointer?

At least for H5, UID registers have special meaning: some registers even encode part of ID in ASCII, so a binary -> hex convert might not be very precise.

[image]

[UID peripheral for f030rc]: https://github.com/embassy-rs/stm32-data-generated/blob/d63f9313acbb7a678631dd41ef82bf76c418b7ca/stm32-metapac/src/chips/stm32f030rc/pac.rs#L142

Luctins, 2024-10-09:
Just stumbled on this today, same thing in the `H563`. 

Maybe a doc comment or a feature gate around this would be in order?

joshuajylin, 2025-01-20:
The same on H563 and casting to [u16;6] is worked.

xoviat, 2025-10-30:
Closed in #4755.