embassy-rs/embassy #2700: nrf52840 please add dcdc voltage setting (UICR.REGOUT0) to config

DeepSOIC, 2024-03-15:
Hi all!
I am programming an nrf52840-dongle. It derives its power from usb using a buck converter in the nrf chip, and each time i flash it, the output voltage setting (UICR.REGOUT0) is reset to 1.8V. I need it running at 3.3V.

So far, i have not found REGOUT0 exposed in the hal anywhere, nor have i found a way to write to UICR in a nice way (well, there is a [`uicr_write_masked`](https://github.com/embassy-rs/embassy/blob/a92a73abe46967a9ca511aa21265b7c338e11c6f/embassy-nrf/src/lib.rs#L362), but it's not exposed).

lulf, 2024-03-21:
Sounds like a good idea, is this something you'd like to contribute?

DeepSOIC, 2024-03-21:
i'm not sure about that... maybe

plaes, 2024-04-01:
This issue can be closed?
cc: @DeepSOIC @Dirbaio

lulf, 2024-04-01:
I believe this can be closed.

closed 2024-04-01 by lulf