embassy-rs/embassy #2702: STM32H7 FDCAN Improper RAM Access

maiaherringfish, 2024-03-15:
I am able to configure both FDCAN1 and FDCAN3, but only FDCAN1 can transmit and receive correctly. 

When I try to transmit with FDCAN3 I am not seeing the actual frame I asked it to write (unlike when I transmit with FDCAN1). Instead I'm seeing the same pretty much nonsense frame written (regardless of whether FDCAN1 is enabled or disabled). 

When I receive and both FDCAN1 and FDCAN3 are enabled and on the same bus, FDCAN1 and FDCAN3 receive the same messages.,but if I disable FDCAN1 I get empty frames on FDCAN3

cschuhen, 2024-03-15:
Yeah, it seems broken for even FDCAN2. I think it may be due to Registers::msg_ram_mut() not taking into account msg_ram_offset.

maiaherringfish, 2024-03-18:
tested your fix @cschuhen and looks like FDCAN3 is working now. thanks for the quick help on this

closed 2024-03-18 by maiaherringfish