embassy-rs/embassy #2704: embassy-executor LoadProhibited exception when using a large task across two different cores in release profile

manuelbaez, 2024-03-16:
Hi, first I'd like to thank anyone who takes the time to read this, sorry about misspellings and any weird things I'm new to Rust and English is my second language :)

I'm trying to make a flight controller with an esp32 using embassy, I started with the esp idf std template but decided to move to no_std out of curiosity and maybe to get more performance and a lighter binary for the app.

 I have my application sectioned into three main tasks at the moment:
- A user input task
- Telemetry/ debugging
- Flight controller

I can run those three tasks just fine if I use only one core, but that introduces some unwanted spikes in the latency for the flight controller and I want it to be as performant as possible, so that's why I want to use the second core exclusively for that. But I'm facing the following issue when trying to use the second core for any of the tasks and also spawn the flight_controller simultaneously:

`Exception occured 'LoadProhibited'
Context
PC=0x400dc1b8       PS=0x00060930
0x400dc1b8 - embassy_executor::raw::util::SyncUnsafeCell<T>::get
    at /mnt/disk-2/repos/embassy/embassy-executor/src/raw/util.rs:55
0x00060930 - ??
    at ??:??`
 
 This only occurs when I try to spawn any of the tasks in the second core, it runs just fine when everything is in just one. At first, I thought it was the way I was sharing the user input data between the tasks, I'm using a struct of atomic values:
 
 `
pub struct AtomicControllerInput {
    pub roll: AtomicI16,
    pub pitch: AtomicI16,
    pub yaw: AtomicI16,
    pub throttle: AtomicU8,
    pub kill_motors: AtomicBool,
    pub start: AtomicBool,
    pub calibrate_esc: AtomicBool,
    pub calibrate_sensors: AtomicBool,
}

impl AtomicControllerInput {
    pub const fn new() -> Self {
        AtomicControllerInput {
            roll: AtomicI16::new(0),
            pitch: AtomicI16::new(0),
            yaw: AtomicI16::new(0),
            throttle: AtomicU8::new(0),
            kill_motors: AtomicBool::new(false),
            start: AtomicBool::new(false),
            calibrate_esc: AtomicBool::new(false),
            calibrate_sensors: AtomicBool::new(false),
        }
    }
}

pub static SHARED_CONTROLLER_INPUT: AtomicControllerInput = AtomicControllerInput::new();
`
So I removed that to test and the executor, where the flight controller is, kept crashing (also I use relaxed ordering, not sure if that's important), so I started to comment code to see at what point it crashes, and I found out it was crashing on an async function that used the I2C but interestingly when I made the function sync it stopped crashing at that point, but when I added some code back it started crashing again.

Then I saw in the docs about the task arena and I enabled the nightly feature, this Increased the amount of my code that I could put back into the function but still crashed at some point on some random arithmetic function, nothing that should cause problems. This is why I think this has something to do with the size of the task( call stack? or Future) since it seems to be trying to access memory out of bounds for the executor.

I decided to clone the embassy repo and tracked down the crash to this function call:

[image]

This is line 405 of the src/raw/mod.rs in the embassy-executor, but  I'm at a loss here since I'm quite new to rust(I picked this project to start learning it) and I don't understand much of what's going on here, but I guess that wherever that poll_fn is getting assigned for that task it's assigning it a pointer that leads to some wrong memory address. 

Important to note, **This only happens in a release build**, somehow debug builds are fine but the code runs 20 to 30x slower so that might be something worth mentioning, no idea why though.
 
 
This is the function that crashes in the executor:
[](https://github.com/manuelbaez/open-drone/blob/chrasing-issue/flight-controller-nostd/src/threads.rs#L42)

Here is the full app code that's crashing:
[](https://github.com/manuelbaez/open-drone/tree/chrasing-issue/flight-controller-nostd)
 
 This is what my main looks like when spawning to the second core:
[... 236 lines cut ...]
	at ~/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1531
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
	at ~/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1531
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
	at ~/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1531
F15=0x00000000
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
	at ~/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1531

0x400d4a47
0x400d4a47 - _ZN7esp_hal3soc14implementation11cpu_control10CpuControl16start_core1_init17h8a56d85e6dac564bE.llvm.534746425162065537
	at ??:??
0x40000000
0x40000000 - _ZN12embassy_time13queue_generic5QUEUE17ha6ddcddb6f3e2583E
	at ??:??
0x40000740
0x40000740 - _ZN12embassy_time13queue_generic5QUEUE17ha6ddcddb6f3e2583E
	at ??:??
0x40000000
0x40000000 - _ZN12embassy_time13queue_generic5QUEUE17ha6ddcddb6f3e2583E
	at ??:??

`

Again thanks in advance for your time and any comments!

MabezDev, 2024-03-21:
I can't see what esp-hal version your using, but ensure it includes https://github.com/esp-rs/esp-hal/pull/1286 if you're using the second core at all. For the original ESP32, you will also want to have https://github.com/esp-rs/esp-hal/pull/1289 included.

manuelbaez, 2024-03-22:
Hey, thanks for your answer! I pulled again from main and it seems to have fixed the issue, also it fixed the I2c frequency that was way off before. The I2c still crashes with AckCheckFailed when used in the second core and the SCL line locks forever, though for my use case, I can just run that task in core 0 as it works fine. I think I'll open an issue in the esp-hal about this later.

This is what I meant about the SCL line:
[image]

closed 2024-03-22 by manuelbaez