embassy-rs/embassy #2705: H743ZI2 not supported ?

npf, 2024-03-18:
Testing examples on H743ZI2 (H743ZI is not produced anymore if I'm correct) using `--chip STM32H743ZITx`, the `usb_serial` fails.

```
$ cargo run --bin usb_serial
   Compiling embassy-stm32h7-examples v0.1.0 (/home/neyron/scm/rust/stm32/embassy/examples/stm32h7)
    Finished dev [optimized + debuginfo] target(s) in 1.17s
     Running `probe-rs run --chip STM32H743ZITx target/thumbv7em-none-eabihf/debug/usb_serial`
      Erasing ✔ [00:00:01] [############################################################] 128.00 KiB/128.00 KiB @ 69.07 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [##############################################################] 57.00 KiB/57.00 KiB @ 57.03 KiB/s (eta 0s )    Finished in 2.864s
 WARN defmt_decoder::log::format: logger format contains timestamp but no timestamp implementation was provided; consider removing the timestamp (`{t}` or `{T}`) from the logger format or provide a `defmt::timestamp!` implementation    
0.000000 INFO  Hello World!
└─ usb_serial::____embassy_main_task::{async_fn#0} @ src/bin/usb_serial.rs:20  
0.000000 DEBUG flash: latency=2 wrhighfreq=2
└─ embassy_stm32::rcc::_version::flash_setup @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:130 
0.000000 TRACE BDCR ok: 00008200
└─ embassy_stm32::rcc::bd::{impl#3}::init @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000000 DEBUG rcc: Clocks { csi: Some(Hertz(4000000)), hclk1: Some(Hertz(200000000)), hclk2: Some(Hertz(200000000)), hclk3: Some(Hertz(200000000)), hclk4: Some(Hertz(200000000)), hse: None, hsi: Some(Hertz(64000000)), hsi48: Some(Hertz(48000000)), i2s_ckin: None, lse: None, lsi: None, pclk1: Some(Hertz(100000000)), pclk1_tim: Some(Hertz(200000000)), pclk2: Some(Hertz(100000000)), pclk2_tim: Some(Hertz(200000000)), pclk3: Some(Hertz(100000000)), pclk4: Some(Hertz(100000000)), pll1_q: None, pll2_p: None, pll2_q: None, pll2_r: None, pll3_p: None, pll3_q: None, pll3_r: None, rtc: Some(Hertz(32000)), sys: Some(Hertz(400000000)) }
└─ embassy_stm32::rcc::set_freqs @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:130 
0.000000 TRACE allocating type=Interrupt mps=8 interval_ms=255, dir=In
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000030 TRACE   index=1
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000030 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=Out
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000061 TRACE   index=1
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000091 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=In
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000091 TRACE   index=2
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000122 INFO  USB: device_descriptor used: 18
└─ embassy_usb::builder::{impl#1}::build @ /home/neyron/scm/rust/stm32/embassy/embassy-usb/src/fmt.rs:143 
0.000122 INFO  USB: config_descriptor used: 70
└─ embassy_usb::builder::{impl#1}::build @ /home/neyron/scm/rust/stm32/embassy/embassy-usb/src/fmt.rs:143 
0.000152 INFO  USB: bos_descriptor used: 12
└─ embassy_usb::builder::{impl#1}::build @ /home/neyron/scm/rust/stm32/embassy/embassy-usb/src/fmt.rs:143 
0.000152 INFO  USB: msos_descriptor used: 0
└─ embassy_usb::builder::{impl#1}::build @ /home/neyron/scm/rust/stm32/embassy/embassy-usb/src/fmt.rs:143 
0.000152 INFO  USB: control_buf size: 64
└─ embassy_usb::builder::{impl#1}::build @ /home/neyron/scm/rust/stm32/embassy/embassy-usb/src/fmt.rs:143 
0.000183 TRACE allocating type=Control mps=64 interval_ms=0, dir=Out
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000213 TRACE   index=0
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000213 TRACE allocating type=Control mps=64 interval_ms=0, dir=In
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000244 TRACE   index=0
└─ embassy_stm32::usb_otg::usb::{impl#6}::alloc_endpoint @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000244 TRACE start
└─ embassy_stm32::usb_otg::usb::{impl#7}::start @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000274 TRACE Core id 00002300
└─ embassy_stm32::usb_otg::usb::{impl#9}::init @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117 
0.000274 TRACE SETUP waiting
└─ embassy_stm32::usb_otg::usb::{impl#18}::setup::{async_fn#0}::{closure#0} @ /home/neyron/scm/rust/stm32/embassy/embassy-stm32/src/fmt.rs:117
```

Trying chip 'STM32H743ZI2Tx', I get `was not found in the list of known targets`.

I see that github.com:embassy-rs/stm32-data does not have `build/data/chips/STM32H743ZI2.json`. Should it be added ?

Dirbaio, 2024-03-18:
the "STM32H743ZI2" chip doesn't exist. There's a "Nucleo-H743ZI2" *board*, which has the STM32H743ZI chip. I'm not sure what the "2" means there but it's not the issue. 

Have you changed the chip in `Cargo.toml` as well? not just `.cargo/config.toml`.

npf, 2024-03-19:
Ah yes, indeed it's a Nucleo-H743ZI2, sorry for my misunderstanding.

I changed the chip in `Cargo.toml` to as you suggested like in `.cargo/config.toml`. However it still does not work: no `/dev/ttyACM1`, no message in `dmesg` and same debug output as above.

Looking for the difference between H743ZI and H743ZI2, I found this thread: https://community.st.com/t5/stm32-mcus-boards-and-hardware/difference-between-stm32h743zi-and-stm32h743zi2/td-p/235754, which reports USB changes?

Dirbaio, 2024-03-19:
Ah, found it. Add this to RCC config and it'll work: `config.rcc.mux.usbsel = mux::Usbsel::HSI48;`

(the issue is [this PR](https://github.com/embassy-rs/embassy/pull/2626/files#diff-41c3a4c6bf9f7a2d119a6cd4c6d01f6d4ce9c1966d3b1d9f46b8167bb3a1a1ea) changed the USB clock mux to be controlled by RCC config, and the default is "disabled" :sweat: . I missed updating the examples, will do soon)

PR #2711 (2024-03-19): stm32/usb: assert correct clock on init, set mux in all examples.
- **stm32/usb: merge usb and usb_otg into single module.**
- **stm32/usb: extract common init code.**
- **stm32/usb: assert usb clock is okay.**
- **stm32/usb: ensure mux is configured in examples.**

fixes #2705
closed 2024-03-19 by Dirbaio