embassy-rs/embassy #2709: embassy-executor-macros fails to compile on the latest Rust Nightly.

danclive, 2024-03-19:
see: https://github.com/esp-rs/esp-hal/actions/runs/8342745449/job/22831570036#step:7:609

```
   Compiling examples v0.0.0 (/home/runner/work/esp-hal/esp-hal/examples)
error: concrete type differs from previous defining opaque type use
  --> src/bin/embassy_usb_serial_jtag.rs:27:1
   |
27 | #[embassy_executor::task]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^ expected `writer::Fut`, got `impl Future<Output = ()>`
   |
note: previous use here
  --> src/bin/embassy_usb_serial_jtag.rs:27:1
   |
27 | #[embassy_executor::task]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^
   = note: this error originates in the attribute macro `embassy_executor::task` (in Nightly builds, run with -Z macro-backtrace for more info)

error: concrete type differs from previous defining opaque type use
  --> src/bin/embassy_usb_serial_jtag.rs:47:1
   |
47 | #[embassy_executor::task]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^ expected `reader::Fut`, got `impl Future<Output = ()>`
   |
note: previous use here
  --> src/bin/embassy_usb_serial_jtag.rs:47:1
   |
47 | #[embassy_executor::task]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^
   = note: this error originates in the attribute macro `embassy_executor::task` (in Nightly builds, run with -Z macro-backtrace for more info)
```

danclive, 2024-03-19:
This may be due to the recent Rust Nightly having a destructive impact on embassy-executor-macros.

https://github.com/embassy-rs/embassy/blob/ea25112f7d1ce27a7216ce6c3aab983dc3a0d7c0/embassy-executor-macros/src/macros/task.rs#L93-L101

jessebraham, 2024-03-20:
I understand that the stated policy here is that only the nightly version in the `rust-toolchain.toml` file is officially supported, but how are HAL developers expected to accommodate this? This is a pretty rigid restriction that I'm not really comfortable applying to our HAL.

Dirbaio, 2024-03-20:
seems a rust regression to me. Filed https://github.com/rust-lang/rust/issues/122775

Seems #2574 is not affected

Dirbaio, 2024-03-20:
fixed in #2574

closed 2024-03-20 by Dirbaio