embassy-rs/embassy #2715: Emmc not directly supportin on STM32

snakehand, 2024-03-20:
The Sdmmc::init_card() function assumes that it is talking to an SD card, but the initialisation procedure for eMMC devices is slightly different. For instance CMD55 is not required prior to sending CMD6 setting the bus width. I will try to address this issue, but I am not sure if SD / eMMC option should be part of the config parameter, or if a new function init_emmc() should be made.

Dirbaio, 2024-03-20:
if the method signature is the same (ie they require the same amount of pins etc) then there's no reason to make a separate method, a config setting is better.

snakehand, 2024-03-21:
Signature is the same, and a modal setting in the configuration is probably the best, as it does not add much cyclomatic complexity, but avoids a certain amount of duplicated code.