embassy-rs/embassy #2723: [STM32] Support LTDC peripheral

JuliDi, 2024-03-21:
It is already (partially) supported in stm32f4xx hal (https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/ltdc/index.html) but embassy does not seem to have any support for this, yet.

I would like to use it with the STM32F4 Discovery kit for STM32F469.
An example for how this could work with, e.g., slint is found here https://github.com/slint-ui/slint/blob/master/examples/mcu-board-support/stm32h735g.rs though this is with an STM32H735G, but I guess it would work very similar.
As you can see in the example, support in the STM32yyxx-hals is not great and I think being able to interface with displays from embassy would be a great addition and make for some nice demos.

Currently I have no good ideas what an implementation would look like. So any input on this is welcome. Eventually I would like to also work on some abstractions over the exact display used, but that will supposedly need some more discussion once we have LTDC confirmed working.

In the meantime, supporting the peripheral on a PAC level from stm32-data would be great and higher level abstractions could be implemented once the PAC part is confirmed working.

joelsa, 2024-03-28:
Hi, I'd like to take a look at this as well!

joelsa, 2024-05-20:
Can be closed since https://github.com/embassy-rs/stm32-data/pull/468 and https://github.com/embassy-rs/embassy/pull/2903 are merged now.

closed 2026-11-05 by leftger