embassy-rs/embassy #2743: Example code errors and version management issues

juebanlin, 2024-03-27:
I have found that many embedded examples of Rust cannot follow the version well and achieve a usable snapshot, just like a manual. You need to know the exact version of the dependency used in this example to run correctly
When I wanted to try some examples, I found that the code of the examples did not change correspondingly with the API changes,
Many modules in the project are interdependent, and the example module is not properly bound to the versions of other modules as an independent version module, resulting in many examples requiring code revisions
Solution 1:
Modify each API while modifying all dependent examples;
Solution 2:
The example code is independent and bound to the versions of other modules, ensuring that examples within one version can be executed correctly
This is just beginners, it's too bad
Can anyone share an example of using I2C to read SMBus? I always timeout when reading, I don't know what else to do.

pkoevesdi, 2024-04-01:
> For those who are new to Rust embedded systems, the experience is too bad.

I want to totally stress this fact. I see not much helpful informations how to use certain modules, and I find masses of examples that just don't work. For instance, I'm struggling with `eth_client.rs` and don't know where to get help, because I don't understand even half of that source. Now, the more I try around, the more I have the feeling, that it's not purely my lack of understanding - it just doesn't work.
I think, providing working examples is a major key to success for a language. Make it easy for people to get running! Then they will have fun in the language and can easily by try-n-error, starting with a working code. You can't if You start off from scratch or from a not-surely working code.

Dirbaio, 2024-04-01:
examples on latest git main correspond to embassy crates from latest git main. They are tested in CI, they're not out of date.

If you want examples that work with, for example, embassy-stm32 v0.1.0 released on crates.io, you have to go back in the repo history from when it was released: https://github.com/embassy-rs/embassy/tree/embassy-stm32-v0.1.0/examples

pkoevesdi, 2024-04-06:
Ok, I get an example (`usb_serial.rs`) working from the cloned embassy repo. That includes several relative paths in `Cargo.toml`. As soon as I delete the `path`s from the dependencies (which presumably makes it take the crates.io packages), I get compile errors, in this case from missing module `usb` inside `embassy-stm32`. I checked all modules from crates.io: they have the same version number as in the embassy repo's `Cargo.toml`s. 
But when I compare the sources from the repo, i.e. `embassy/embassy-stm32/src` with the ones from crates.io, i.e. `.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-stm32-0.1.0/src`, there are many differences.

I'd like to understand, how come that they are different, when having the same version number, down to the minor?

And why is there no module `usb` in the crates.io version of `embassy-stm32` (at least according to the compiler message)? In https://docs.embassy.dev/embassy-stm32/git/stm32h730vb/usb/index.html there it is.

Dirbaio, 2024-04-06:
The version number in git is bumped right before release. So embassy-stm32 on git still says v0.1.0 but has breaking changes. It will be released as v0.2.0 when time comes.

As I said, if you want examples that work with the crates.io released embassy-stm32 v0.1.0, grab them from further back in the repo history, here https://github.com/embassy-rs/embassy/tree/embassy-stm32-v0.1.0/examples

pkoevesdi, 2024-04-06:
Ok, thanks for the explanation. I use `git = "https://github.com/embassy-rs/embassy.git"` inside my project's `Cargo.toml` dependencies for now, for not needing to link into a local copy of embassy.

pkoevesdi, 2024-04-07:
> And why is there no module usb in the crates.io version of embassy-stm32 (at least according to the compiler message)? In [docs.embassy.dev/embassy-stm32/git/stm32h730vb/usb/index.html](https://docs.embassy.dev/embassy-stm32/git/stm32h730vb/usb/index.html) there it is.

For my own reference: Here's the diff which makes it clear:
[`embassy-stm32-v0.1.0...main`#diff-ccb91ef3e4](https://github.com/embassy-rs/embassy/compare/embassy-stm32-v0.1.0...main#diff-ccb91ef3e4d33e0337df2652d5be85fb8c736e67396994765422fe5ef7c5cf17)

closed 2024-04-26 by Dirbaio