embassy-rs/embassy #2753: embassy-usb: pass larger reads and writes to the driver

simpkins, 2024-03-30:
The embassy-usb code currently handles breaking up large endpoint reads and writes into individual packets in the higher level code.  (Each user of the EndpointIn and EndpointOut APIs appears to be responsible for doing this on their own.)

It would be nice to move this functionality down into the embassy-usb-driver implementations instead, as currently this seems like it will make reads and writes more inefficient than necessary, requiring the hardware to NAK tokens from the host in between each packet, until the higher-level calling code provides the next packet.

e.g., the Synopsys USB cores used by ESP32 and STM32 devices can accept larger read and write transactions on its own: you can tell it you are about to start a multi-packet transaction and it can then send/receive multiple packets from the host without needing to wait for any subsequent intervention from the driver code.  Currently in the ESP32 and STM32 embassy-driver-usb implementations we can't take advantage of the full FIFO space available in the chips since we can only send or receive at most one packet per endpoint at a time.

There are some minor edge cases that would need to be addressed in the API design, but I think some of these exist in the current API.  In particular: if you call `EndpointOut::read()`, and then drop the read operation (say, due to a timeout), what happens to any partially read data?  (i.e., is there a mechanism to inform the caller how much data was placed into their buffer before the operation was dropped, and do we guarantee that no data was lost if the user restarts a read later.)  Dealing with dropped reads seems like a slightly tricky corner case even now if we want to guarantee that no data is lost when the read is dropped, and will be returned by the next read instead.

Dirbaio, 2024-03-30:
Yes, delegating transfer chunking into packets to the driver sounds good to me. It'd be nice to let the hardware take care of that on peripherals that support it.

(it's also annoying that each user has to replicate the chukning code, including the ZST and all)

muness, 2026-10-06:
Following up: the `read_transfer`/`write_transfer` hooks fixed the API side of this, but `embassy-usb-synopsys-otg` still used the default packet-by-packet implementation. So on ESP32/STM32 the core still NAKed the host between every OUT packet, the inefficiency described above. #7191 overrides `read_transfer` in the Synopsys driver to arm multi-packet OUT transfers (opt-in via `Config::bulk_out_transfer_bytes`). On an ESP32-S3 at full speed, CDC-NCM OUT goes from 4.54 to 8.41 Mbit/s. Reviews or an STM32 test run would be very welcome.