embassy-rs/embassy #2770: docs: Is there any plan for i18n?

decaday, 2024-04-04:
I have translated most of the chapters of Embassy docs into Simplified Chinese, which is currently on my [github pages](https://decaday.github.io/embassy-docs-zh/zh/index.html).
And I plan to continue to maintain the Simplified Chinese translation.

If there are plans for i18n, there are some things to consider
For example, whether to use tools such as po4a, or directly maintain documents in different languages separately, like Wikipedia.
I found [antora's ui for i18n](https://gitlab.com/antora-ui-custom/antora-i18n/-/blob/main/docs/en/modules/ROOT/pages/readme.adoc?ref_type=heads). To use it, we may also need to reorganize the directory structure

Some other methods are mentioned at https://coastsystems.net/en/blog/2022-01-04-antora/.
I'm new to antora and don't know if there is a better way.

thank you for your reply!

lulf, 2024-04-04:
I think you're the first to think about i18n to be honest :) I think reorganizing the dir structure should be doable, like moving the current docs/* into docs/en/* or something like that?

decaday, 2024-04-05:
> I think you're the first to think about i18n to be honest :) I think reorganizing the dir structure should be doable, like moving the current docs/* into docs/en/* or something like that?

Thank you!

You'd rather use a translation management tool like po4a (like [this](https://gitlab.com/antora-ui-custom/antora-i18n))
or just maintain documentation in different languages separately (like Wikipedia)?

Additionally, both methods appear to take up the `version` key. The current `version` key of Embassy is `dev`.
I couldn't find a way to avoid occupying the version key...

lulf, 2024-04-05:
I have no experience with po4a, so I'm not sure how well it translates the doc, would you able to try it and make a judgement of how well it works?

I think for now, using language instead of version key could be ok. Having different docs per version might be relevant some day, but maybe Antora will have better support for it in the future.

decaday, 2024-04-05:
Sure, I will be making a presentation and giving an evaluation in my fork soon.

I will not be at home for the next few days, and I will complete this attempt next weekend.