embassy-rs/embassy #2772: note: rust-lld: error: undefined symbol: _defmt_panic

gustavowd, 2024-04-04:
I'm trying to start a embassy project for a STM32G4 microcontroller from scratch, but i'm facing an persistent error.
When i try to compile with cargo build or cargo embed i'm getting the following error:

`a1af864f49" "--gc-sections" "-Tlink.x"
  = note: rust-lld: error: undefined symbol: _defmt_panic
          >>> referenced by mod.rs:133 (/home/gustavo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.6/src/export/mod.rs:133)
          >>>               /home/gustavo/embedded_rust/blinky/target/thumbv7em-none-eabihf/debug/deps/blinky-24951ba1af864f49.1e549mar4vimxyhk.rcgu.o:(embassy_executor::_export::Arena$LT$_$GT$::alloc::hdab4712089b2b616)
          >>> referenced by mod.rs:133 (/home/gustavo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.6/src/export/mod.rs:133)
          >>>               /home/gustavo/embedded_rust/blinky/target/thumbv7em-none-eabihf/debug/deps/blinky-24951ba1af864f49.cprwdx9gysemz3b.rcgu.o:(embassy_executor::spawner::Spawner::must_spawn::ha10faf6f94b6545b)
          >>> referenced by mod.rs:133 (/home/gustavo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.6/src/export/mod.rs:133)
          >>>               embassy_stm32-16eff0fd6274d331.embassy_stm32.f6033aee96536bb7-cgu.2.rcgu.o:(embassy_stm32::_generated::Peripherals::take_with_cs::h7e879cecd6fa453d) in archive /home/gustavo/embedded_rust/blinky/target/thumbv7em-none-eabihf/debug/deps/libembassy_stm32-16eff0fd6274d331.rlib
          >>> referenced 18 more times
          
          rust-lld: error: section .defmt.{"package":"blinky","tag":"defmt_info","data":"Hello World!","disambiguator":"14157196540590754686","crate_name":"blinky"} virtual address range overlaps with .text
          >>> .defmt.{"package":"blinky","tag":"defmt_info","data":"Hello World!","disambiguator":"14157196540590754686","crate_name":"blinky"} range is [0x80001D8, 0x80001D8]
          >>> .text range is [0x80001D8, 0x800EC53]
          
          rust-lld: error: section .text virtual address range overlaps with .defmt.{"package":"blinky","tag":"defmt_info","data":"high","disambiguator":"5785278663893197478","crate_name":"blinky"}
          >>> .text range is [0x80001D8, 0x800EC53]
          >>> .defmt.{"package":"blinky","tag":"defmt_info","data":"high","disambiguator":"5785278663893197478","crate_name":"blinky"} range is [0x80001D9, 0x80001D9]
          
          rust-lld: error: section .defmt.{"package":"blinky","tag":"defmt_info","data":"Hello World!","disambiguator":"14157196540590754686","crate_name":"blinky"} load address range overlaps with .text
          >>> .defmt.{"package":"blinky","tag":"defmt_info","data":"Hello World!","disambiguator":"14157196540590754686","crate_name":"blinky"} range is [0x80001D8, 0x80001D8]
          >>> .text range is [0x80001D8, 0x800EC53]
          
          rust-lld: error: section .text load address range overlaps with .defmt.{"package":"blinky","tag":"defmt_info","data":"high","disambiguator":"5785278663893197478","crate_name":"blinky"}
          >>> .text range is [0x80001D8, 0x800EC53]
          >>> .defmt.{"package":"blinky","tag":"defmt_info","data":"high","disambiguator":"5785278663893197478","crate_name":"blinky"} range is [0x80001D9, 0x80001D9]
          

error: aborting due to 1 previous error

       Error Failed to run cargo build: exit code = Some(101).`

[main.rs]
`use defmt::*;
use embassy_executor::Spawner;
use embassy_stm32::gpio::{Level, Output, Speed};
use embassy_time::Timer;
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");

    let mut led = Output::new(p.PA5, Level::High, Speed::Low);

    loop {
        info!("high");
        led.set_high();
        Timer::after_millis(300).await;

        info!("low");
        led.set_low();
        Timer::after_millis(300).await;
    }
}`

[cargo.toml]
`[package]
name = "blinky"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
cortex-m = { version = "0.7.7", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.3"

panic-probe = { version = "0.3.1", features = ["print-defmt"] }
futures = { version = "0.3.17", default-features = false, features = ["async-await"] }
heapless = { version = "0.8", default-features = false }
static_cell = "2.0.0"

embassy-stm32 = {version = "0.1.0", features = [ "defmt", "time-driver-any", "stm32g474re", "memory-x", "unstable-pac", "exti"] }
embassy-sync = {version = "0.5.0", features = ["defmt"]}
embassy-executor = {version = "0.5.0", features = ["arch-cortex-m", "executor-thread", "defmt", "integrated-timers"]}
embassy-time = {version = "0.3.0", features = ["defmt", "defmt-timestamp-uptime", "tick-hz-32_768"]}
embassy-futures = "0.1.0"
defmt = "0.3.6"
defmt-rtt = "0.4"

[profile.release]
debug = 2`

[config.toml]
`[build]
target = "thumbv7em-none-eabihf"

[target.thumbv7em-none-eabihf]
#rustflags = ["-C", "link-arg=-Tdefmt.x"]
rustflags = ["-C", "link-arg=-Tlink.x"]

#[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# replace STM32G071C8Rx with your chip as listed in `probe-rs chip list`
#runner = "probe-rs run --chip STM32G474RETx"

[env]
DEFMT_LOG = "trace"`

Any ideais?
Thanks in advance

gustavowd, 2024-04-04:
The first one i was able to sove using:

#[defmt::panic_handler]
fn panic() -> ! {
    core::panic!("panic via `defmt::panic!`")
}

closed 2024-04-04 by gustavowd

Dirbaio, 2024-04-04:
you're missing `-Tdefmt.x` linker flag. 

I suggest adding them via build.rs like the examples do, not via .cargo/config.toml.

gustavowd, 2024-04-05:
Thanks @Dirbaio .

Just one more question. Why when inserting the build.rs file I no longer need to have a memory.x file? In fact, when maintaining the memory.x file I receive a message that the FLASH memory has already been declared elsewhere.

Dirbaio, 2024-04-05:
`memory-x` feature in `embassy-stm32` automatically provides a memory.x file. if you want to provide your own, you have to disble it.