embassy-rs/embassy #2773: defmt printing wrong characters

gustavowd, 2024-04-04:
I'm trying to start a embassy project for a STM32G4 microcontroller from scratch, but i'm facing an persistent error.
When i use cargo embed to flash and run the code, i'm getting wrong characters for info! macros.

The code is:

`use embassy_executor::Spawner;
use embassy_stm32::gpio::{Level, Output, Speed};
use embassy_time::Timer;
use defmt::*;
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");
    //defmt::println!("Hello, world!");

    let mut led = Output::new(p.PA5, Level::High, Speed::Low);

    loop {
        info!("high");
        led.set_high();
        Timer::after_millis(500).await;

        info!("low");
        led.set_low();
        Timer::after_millis(500).await;
    }
}`

And this code is printing ...
` defmt                                                                                                                               
17:03:25.414 ~�7~
17:03:25.414 W$�i$�$4�$�$�M$�&$�S$�i6K}mzz�r��bKbj�bb8&br�-bʀ5b
`

What could i possible do in this case?

Dirbaio, 2024-04-04:
Install the latest probe-rs version following the instructions at https://probe.rs/ . This comes with a newer version of cargo-embed that supports defmt.  Alternatively you can run with `probe-rs run`, like the examples in this repo do.

gustavowd, 2024-04-05:
Thanks @Dirbaio. It is working now with probe-rs run.
cargo-embed still shows me dumb characters.

closed 2024-04-05 by gustavowd