embassy-rs/embassy #2780: Couldn't run usb_serial.rs example on a stm32f4

silvestrpredko, 2024-04-06:
Hey, I couldn't successfully run the usb_serial example, the system hangs after the config was set up.

Basically
```Rust
info!("Hello World!");

let mut config = Config::default();
{
    use embassy_stm32::rcc::*;
    config.rcc.hse = Some(Hse {
        freq: Hertz(8_000_000),
        mode: HseMode::Bypass,
    });
    config.rcc.pll_src = PllSource::HSE;
    config.rcc.pll = Some(Pll {
        prediv: PllPreDiv::DIV4,
        mul: PllMul::MUL168,
        divp: Some(PllPDiv::DIV2), // 8mhz / 4 * 168 / 2 = 168Mhz.
        divq: Some(PllQDiv::DIV7), // 8mhz / 4 * 168 / 7 = 48Mhz.
        divr: None,
    });
    config.rcc.ahb_pre = AHBPrescaler::DIV1;
    config.rcc.apb1_pre = APBPrescaler::DIV4;
    config.rcc.apb2_pre = APBPrescaler::DIV2;
    config.rcc.sys = Sysclk::PLL1_P;
    config.rcc.mux.clk48sel = mux::Clk48sel::PLL1_Q;
}
let p = embassy_stm32::init(config);
info!("Hello World! 2"); // => this line never printed
```

atmelmicro, 2024-04-06:
Hey, I had a similiar issue, changing the HseMode::Bypass to HseMode::Oscillator will get you further (most st dev boards have a oscillator, not a clock). However for me it still isn't recognised as a device in windows or macos.

silvestrpredko, 2024-04-06:
> Hey, I had a similiar issue, changing the HseMode::Bypass to HseMode::Oscillator will get you further (most st dev boards have a oscillator, not a clock). However for me it still isn't recognised as a device in windows or macos.

Yeah, it still doesn't recognize as a USB device :(.

atmelmicro, 2024-04-06:
update: for the 32F429IDISCOVERY you need to use the HS interface (even for FS). No clue why it's wired that way
[image]
(PB15 and PB14) are HS pins, but only wired for FS afaik

```rs    
bind_interrupts!(struct Irqs {
    OTG_HS => usb::InterruptHandler<peripherals::USB_OTG_HS>;
});

let driver = Driver::new_fs(p.USB_OTG_HS, Irqs, p.PB15, p.PB14, &mut ep_out_buffer, config);    
```

silvestrpredko, 2024-04-06:
> update: for the 32F429IDISCOVERY you need to use the HS interface (even for FS). No clue why it's wired that way
>
> [image]
>
> (PB15 and PB14) are HS pins, but only wired for FS afaik
>
>
>
> ```rs [...]

Oh let me try, I will respond on Monday.

Thanks.

atmelmicro, 2024-04-06:
This makes the device show up, but it throws a code 43. On macOS it doesn't show up at all. 
[image]
I'll post an update if I figure out something

atmelmicro, 2024-04-06:
logs for reference

```
 WARN defmt_decoder::log::format: logger format contains timestamp but no timestamp implementation was provided; consider removing the timestamp (`{t}` or `{T}`) from the logger format or provide a `defmt::timestamp!` implementation    
0.000000 INFO  Hello World!
└─ usb_serial::____embassy_main_task::{async_fn#0} @ src\bin\usb_serial.rs:21  
0.000000 TRACE BDCR ok: 00008200
└─ embassy_stm32::rcc::bd::{impl#2}::init @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117 
0.000000 DEBUG flash: latency=5
└─ embassy_stm32::rcc::_version::init @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:130 
0.000000 DEBUG rcc: Clocks { hclk1: Some(Hertz(168000000)), hclk2: Some(Hertz(168000000)), hclk3: Some(Hertz(168000000)), pclk1: Some(Hertz(42000000)), pclk1_tim: Some(Hertz(84000000)), pclk2: Some(Hertz(84000000)), pclk2_tim: Some(Hertz(168000000)), pll1_q: Some(Hertz(48000000)), pllsai1_q: None, rtc: Some(Hertz(32000)), sys: Some(Hertz(168000000)) }
└─ embassy_stm32::rcc::set_freqs @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:130 
0.000000 INFO  inited
└─ usb_serial::____embassy_main_task::{async_fn#0} @ src\bin\usb_serial.rs:45  
0.000183 TRACE allocating type=Interrupt mps=8 interval_ms=255, dir=In
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117 
0.000427 TRACE   index=1
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117 
0.000579 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=Out
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117 
0.000823 TRACE   index=1
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117 
0.000946 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=In
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117 
0.001220 TRACE   index=2
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117 
0.001373 INFO  USB: config_descriptor used: 70
└─ embassy_usb::builder::{impl#1}::build @ C:\Users\David\Documents\new\embassy2\embassy-usb\src\fmt.rs:143 
0.001464 INFO  USB: bos_descriptor used: 12
└─ embassy_usb::builder::{impl#1}::build @ C:\Users\David\Documents\new\embassy2\embassy-usb\src\fmt.rs:143 
0.001586 INFO  USB: msos_descriptor used: 0
└─ embassy_usb::builder::{impl#1}::build @ C:\Users\David\Documents\new\embassy2\embassy-usb\src\fmt.rs:143
0.001708 INFO  USB: control_buf size: 64
└─ embassy_usb::builder::{impl#1}::build @ C:\Users\David\Documents\new\embassy2\embassy-usb\src\fmt.rs:143
0.001831 TRACE allocating type=Control mps=64 interval_ms=0, dir=Out
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.002075 TRACE   index=0
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.002197 TRACE allocating type=Control mps=64 interval_ms=0, dir=In
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.002441 TRACE   index=0
└─ embassy_stm32::usb::_version::{impl#6}::alloc_endpoint @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.002563 TRACE start
└─ embassy_stm32::usb::_version::{impl#7}::start @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.002655 INFO  Starting echo
└─ usb_serial::____embassy_main_task::{async_fn#0} @ src\bin\usb_serial.rs:93
0.002807 TRACE Core id 00001100
└─ embassy_stm32::usb::_version::{impl#9}::init @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.002929 TRACE irq
└─ embassy_stm32::usb::_version::{impl#0}::on_interrupt @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.003021 TRACE vbus detected
└─ embassy_stm32::usb::_version::{impl#10}::poll::{async_fn#0}::{closure#0} @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.003112 TRACE usb: power detected
└─ embassy_usb::{impl#2}::handle_bus_event::{async_fn#0} @ C:\Users\David\Documents\new\embassy2\embassy-usb\src\fmt.rs:117
0.003204 TRACE enable
└─ embassy_stm32::usb::_version::{impl#10}::enable::{async_fn#0} @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117
0.003295 TRACE SETUP waiting
└─ embassy_stm32::usb::_version::{impl#18}::setup::{async_fn#0}::{closure#0} @ C:\Users\David\Documents\new\embassy2\embassy-stm32\src\fmt.rs:117

```

silvestrpredko, 2024-04-06:
> logs for reference
>
>
>
> ```
>
>  WARN defmt_decoder::log::format: logger format contains timestamp but no timestamp implementation was provided; consider removing the timestamp (`{t}` or `{T}`) from the [...]

I don't know where to dig in that circumstances. 😞

silvestrpredko, 2024-04-17:
> Hey, with the newest embassy commit (from main) it seems to be working. Try it for yourself, if it works you can close this issue.
> 
> lsusb (1st run is with it unpluged, 2nd is with it plugged in) <img alt="Snímek obrazovky 2024-04-17 v 15 50 50" width="682" src="https://private-user-images.githubusercontent.com/114749286/323243830-d386eeb3-f021-4d81-bdbd-91def1840f22.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNjU2MzIsIm5iZiI6MTcxMzM2NTMzMiwicGF0aCI6Ii8xMTQ3NDkyODYvMzIzMjQzODMwLWQzODZlZWIzLWYwMjEtNGQ4MS1iZGJkLTkxZGVmMTg0MGYyMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QxNDQ4NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMTY4NmMxNWQ0YjJkMGIxZjAyNzQ5ZDUzZmU5NDRiMzQ0NzBiNDVjNDdlNTNiZGNmOWM3MTc5ZGVkMzA2ZGZmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.B8xOv242DxnGSzXIAEKyeyu4ApSJuY7jFY7uvMF_XUQ">
> 
> Edited example code for rerefence
> 
> ```rust
> #![no_std]
> #![no_main]
> 
> use defmt::{panic, *};
> use embassy_executor::Spawner;
> use embassy_stm32::time::Hertz;
> use embassy_stm32::usb::{Driver, Instance};
> use embassy_stm32::{bind_interrupts, peripherals, usb, Config};
> use embassy_usb::class::cdc_acm::{CdcAcmClass, State};
> use embassy_usb::driver::EndpointError;
> use embassy_usb::Builder;
> use futures::future::join;
> use {defmt_rtt as _, panic_probe as _};
> 
> bind_interrupts!(struct Irqs {
>     OTG_HS => usb::InterruptHandler<peripherals::USB_OTG_HS>;
> });
> 
> #[embassy_executor::main]
> async fn main(_spawner: Spawner) {
>     info!("Hello World!");
> 
>     let mut config = Config::default();
>     {
>         use embassy_stm32::rcc::*;
>         config.rcc.hse = Some(Hse {
>             freq: Hertz(8_000_000),
>             mode: HseMode::Oscillator,
>         });
>         config.rcc.pll_src = PllSource::HSE;
>         config.rcc.pll = Some(Pll {
>             prediv: PllPreDiv::DIV4,
>             mul: PllMul::MUL168,
>             divp: Some(PllPDiv::DIV2), // 8mhz / 4 * 168 / 2 = 168Mhz.
>             divq: Some(PllQDiv::DIV7), // 8mhz / 4 * 168 / 7 = 48Mhz.
>             divr: None,
>         });
>         config.rcc.ahb_pre = AHBPrescaler::DIV1;
>         config.rcc.apb1_pre = APBPrescaler::DIV4;
>         config.rcc.apb2_pre = APBPrescaler::DIV2;
>         config.rcc.sys = Sysclk::PLL1_P;
>         config.rcc.mux.clk48sel = mux::Clk48sel::PLL1_Q;
>     }
>     let p = embassy_stm32::init(config);
> 
>     // Create the driver, from the HAL.
>     let mut ep_out_buffer = [0u8; 256];
>     let mut config = embassy_stm32::usb::Config::default();
>     config.vbus_detection = true;
>     let driver = Driver::new_fs(p.USB_OTG_HS, Irqs, p.PB15, p.PB14, &mut ep_out_buffer, config);
> 
>     // Create embassy-usb Config
>     let mut config = embassy_usb::Config::new(0xc0de, 0xcafe);
>     config.manufacturer = Some("Embassy");
>     config.product = Some("USB-serial example");
>     config.serial_number = Some("12345678");
> 
>     // Required for windows compatibility.
>     // https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.9.1/kconfig/CONFIG_CDC_ACM_IAD.html#help
>     config.device_class = 0xEF;
>     config.device_sub_class = 0x02;
>     config.device_protocol = 0x01;
>     config.composite_with_iads = true;
> 
>     // Create embassy-usb DeviceBuilder using the driver and config.
>     // It needs some buffers for building the descriptors.
>     let mut config_descriptor = [0; 256];
>     let mut bos_descriptor = [0; 256];
>     let mut control_buf = [0; 64];
> 
>     let mut state = State::new();
> 
>     let mut builder = Builder::new(
>         driver,
>         config,
>         &mut config_descriptor,
>         &mut bos_descriptor,
>         &mut [], // no msos descriptors
>         &mut control_buf,
>     );
> 
>     // Create classes on the builder.
>     let mut class = CdcAcmClass::new(&mut builder, &mut state, 64);
> 
>     // Build the builder.
>     let mut usb = builder.build();
> 
>     // Run the USB device.
>     let usb_fut = usb.run();
> 
>     // Do stuff with the class!
>     let echo_fut = async {
>         loop {
>             class.wait_connection().await;
>             info!("Connected");
>             let _ = echo(&mut class).await;
>             info!("Disconnected");
>         }
>     };
> 
>     // Run everything concurrently.
>     // If we had made everything `'static` above instead, we could do this using separate tasks instead.
>     join(usb_fut, echo_fut).await;
> }
> 
> struct Disconnected {}
> 
> impl From<EndpointError> for Disconnected {
>     fn from(val: EndpointError) -> Self {
>         match val {
>             EndpointError::BufferOverflow => panic!("Buffer overflow"),
>             EndpointError::Disabled => Disconnected {},
>         }
>     }
> }
> 
> async fn echo<'d, T: Instance + 'd>(class: &mut CdcAcmClass<'d, Driver<'d, T>>) -> Result<(), Disconnected> {
>     let mut buf = [0; 64];
>     loop {
>         let n = class.read_packet(&mut buf).await?;
>         let data = &buf[..n];
>         info!("data: {:x}", data);
>         class.write_packet(data).await?;
>     }
> }
> ```

Unfortunately, I don't have this board on hand. Let's go ahead and close the issue if you have tried and it works fine.

closed 2024-04-17 by silvestrpredko