embassy-rs/embassy #2784: rp2040 `SetDutyCycle` is not implemented for pwm

mattiekat, 2024-04-06:
Found a bit of a rough edge when trying to use a motor controller library and found that [`embedded_hal::pwm::SetDutyCycle`](https://docs.rs/embedded-hal/latest/embedded_hal/pwm/trait.SetDutyCycle.html) is not implemented for any of the pwm types.

Looking at the `rp2040-hal` crate I see it was implemented for [rp2040_hal::pwm::Channel](https://docs.rs/rp2040-hal/latest/rp2040_hal/pwm/struct.Channel.html) maybe I am misunderstanding and this should be wrapped instead?

alxpettit, 2024-04-19:
Yeah, I'm super confused how to actually use PWM since we don't have a `set_duty` :c

cylewitruk, 2024-04-21:
Just ran into this as well trying to use an RGB LED module...

alxpettit, 2024-04-22:
Yeah, it's weird. The docs have this to say about `compare`:

```
    /// The output on channel A goes high when `compare_a` is higher than the
    /// counter. A compare of 0 will produce an always low output, while a
    /// compare of `top + 1` will produce an always high output.
    pub compare_a: u16,
    /// The output on channel B goes high when `compare_b` is higher than the
    /// counter. A compare of 0 will produce an always low output, while a
    /// compare of `top + 1` will produce an always high output.
    pub compare_b: u16,
```

It also says at the top:

```
/// Note the period in clock cycles of a slice can be computed as:
/// `(top + 1) * (phase_correct ? 1 : 2) * divider`
```

This was super confusing to me, as it seems from the latter comment that `compare_a` & `compare_b` ought not to come into the period at all. I think what's happening is we're adjusting one clock, and if that clock value is higher than _another_ clock, the pin will rise? I don't exactly get it, and unfortunately the actual code seems to be making use of low-level MCU features with a lot of magic numbers -- so it's hard to infer from that too.

cylewitruk, 2024-04-22:
Yeah I would have expected an implementation more like the [NRF](https://github.com/embassy-rs/embassy/blob/a5b006268b7a20ff4ea6ad0d9090c6a6f2bb4602/embassy-nrf/src/pwm.rs#L586) & [STM32](https://github.com/embassy-rs/embassy/blob/a5b006268b7a20ff4ea6ad0d9090c6a6f2bb4602/embassy-stm32/src/timer/simple_pwm.rs#L61) impls...

I can't seem to get the example for RP to work regardless (Pico W) :shrug:

Dirbaio, 2024-04-22:
current status is the API is low-level, you pass in the values that are directly configured in the hardware. The datasheet explains the meaning of those values.

I agree it's not the most intuitive thing, if someone wants to work on adding a higher-level API where you set the frequency and duty cycle and it does the calculations for you.

cylewitruk, 2024-04-22:
> current status is the API is low-level, you pass in the values that are directly configured in the hardware. The datasheet explains the meaning of those values.
> 
> I agree it's not the most intuitive thing, if someone wants to work on adding a higher-level API where you set the frequency and duty cycle and it does the calculations for you, a pull request would be welcome.

Thanks for the quick confirmation :pray:

cylewitruk, 2024-04-23:
@mattiekat @crabdancing have a peek at #2855 -- it's a start anyway..