embassy-rs/embassy #2788: Add esp32 examples

Firstyear, 2024-04-08:
Hi there,

The project readme claims:

  - <a href="https://github.com/esp-rs">esp-rs</a>, for the Espressif Systems ESP32 series of chips.
    - Embassy HAL support for Espressif chips is being developed in the [esp-rs/esp-hal](https://github.com/esp-rs/esp-hal) repository.
    - Async WiFi, Bluetooth and ESP-NOW is being developed in the [esp-rs/esp-wifi](https://github.com/esp-rs/esp-wifi) repository.

However, there are no examples of using esp-rs in this repository, and esp-rs make no mention of anything related to embassy or any examples in their ecosystem. The best I can find is a single example hidden in the esp wifi project. https://github.com/esp-rs/esp-wifi/blob/b18ed4a4f18dddc8ffe6f9c7e426a270c010024a/esp-wifi/examples/embassy_ble.rs#L17 

Would it be possible add esp-rs examples?

lulf, 2024-04-08:
There are plenty of examples with embassy for esp-rs in the esp-hal repo https://github.com/esp-rs/esp-hal/tree/main/examples/src/bin (anything prefixed with embassy).

Adding examples to embassy means additional hardware that must be tested, verified, documented and ... maintained, and since esp-rs best knows ESP and is able to keep it up2date I think that's a better place for it.

BirdLogics, 2024-08-07:
> There are plenty of examples with embassy for esp-rs in the esp-hal repo https://github.com/esp-rs/esp-hal/tree/main/examples/src/bin (anything prefixed with embassy).
>
> Adding examples to embassy [...]

Would be helpful to add this link to a README in the examples folder then? This issue is the first result to pop up when I search "emabssy esp examples," but I wouldn't have needed to search at all if the esp-hal examples were linked.