embassy-rs/embassy #2789: STM32 - USART - Non Blocking Read

Ralim, 2024-04-08:
Hello,

I've noticed that all of the USART drivers implement the embedded hal for `read()` which specifies that they should block if there is no pending data. For my current application I would be able to reduce logic complexity if I could poll these or only read the pending data rather than having to use separate tasks that may block on a read.

Would there be any room in the future to have some form of API to provide either a non-blocking read (just read out anything pending) or some way to check there is pending data (so can avoid a read if no data)?

Dirbaio, 2024-04-08:
BufferedUart / RingbufferedUart do exactly that, `.read()` immediately returns the available data without having to wait for the buffer to fill up.

Ralim, 2024-04-08:
Hia,

What I'm referring to is these lines
[ringbuffered.read](https://github.com/embassy-rs/embassy/blob/029636e6fc06b484898c5935957b33b60a9c88b2/embassy-stm32/src/usart/ringbuffered.rs#L147)

[buffered.read](https://github.com/embassy-rs/embassy/blob/029636e6fc06b484898c5935957b33b60a9c88b2/embassy-stm32/src/usart/buffered.rs#L309)

Where if there is no data pending, it blocks for _some_ data.

I'm looking for a way to read _only_ the pending without blocking for following data.

phycrax, 2024-04-17:
I think it would be possible to use [poll_once](https://docs.embassy.dev/embassy-futures/git/default/fn.poll_once.html) for your use case.

Dillonmcewan, 2024-05-16:
I ran into a similar issue. We're transitioning from a sync implementation using the stm32-rs crates. We were able to hack a solution using [block_on](https://docs.embassy.dev/embassy-futures/git/default/fn.block_on.html) (I'm guessing poll_once would probably also work), so we didn't block execution while waiting for data. It seems like the correct way to do this with the embedded_io traits would be to have the various USART structs implement [ReadReady](https://docs.rs/embedded-io/0.6.1/embedded_io/trait.ReadReady.html). That way you could do something like:
```
let ready = rx.read_ready()?;
if ready {
    rx.read(&mut rx_buf)?;
}
```
```

kpfleming, 2025-05-25:
`ReadReady` is now available, although it's not an async function so you have to poll it.