embassy-rs/embassy #2796: avoid unnecessary polling in select()

simpkins, 2024-04-09:
I was testing some code and noticed that futures were being polled a lot more frequently than I expected.  It looks like this was due to my use of the `select*()` functions: the `select*()` functions poll all futures any time any one of them wakes up.  If you end up with a "tree" of selects, where one branch of a select call ends up making additional select calls, then the entire tree of futures is polled any time any one of them is woken up.  It seems like the same issue probably exists for `join()`

Fixing this seems like it would probably be complicated, but I figured I would open an issue just to explore possible options.  I believe that select in other async frameworks like tokio don't have this problem.

It seems like in order to fix this we would need a few different things:
- I assume select() would need to explicitly cooperate with the current executor to schedule the children futures basically as separate tasks.  This means select() would need to be able to get the current executor.  I assume we could keep a reference to the current executor in a global variable, and have each Executor update this variable when it starts running, and restore it when it returns.  On multi-core chips like esp32 we could have an array indexed by CPU ID rather than just 1 global variable.
- The select() functions would need (pinned) local variables to keep task headers / task storage for each of the children tasks for them to use while they are running.  I see comments in the executor code that TaskStorage objects must live forever, but I didn't really see exactly why this needs to be true.  If we had TaskStorage objects stored on the stack we certainly would need to implement Drop for them, to remove them from the run queue when they are dropped, but I'm not sure if there are other challenges that would make this impossible.
- Functions like block_on() might need to be made smarter, and have more of a real executor that can track a run queue.  Alternatively we could still have select() fall back to the naive implementation if there is no current executor set in the global variable.

Dirbaio, 2024-04-09:
> I believe that select in other async frameworks like tokio don't have this problem.

They do, join/select works like this everywhere.

> I see comments in the executor code that TaskStorage objects must live forever, but I didn't really see exactly why this needs to be true

Wakers contain a pointer to the TaskStorage, and wakers can live forever. Deallocating a TaskStorage would leave dangling pointers.

(Other runtimes refcount the task storage, deallocating it when the task has exited AND all wakers pointing to it have been dropped. We can't do that because we don't have alloc).

Dirbaio, 2024-04-09:
What you're describing is sort of how [FuturesUnordered](https://docs.rs/futures/latest/futures/stream/struct.FuturesUnordered.html) works. It's sort of a "sub-executor", it creates separate wakers for each "sub-task", so it knows which one needs polling when polled by the "parent" executor.

I don't think something like this is doable without `alloc` though.

simpkins, 2024-04-09:
> They do, join/select works like this everywhere.

I had tried testing tokio with the following code, and didn't see this behavior, but maybe I am misunderstanding and didn't get it right it somehow:
https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=00831a8b17dacef22bd5113b06974c19

With equivalent code in embassy I see task1 and task2 woken up every time the periodic task wakes up.

> Wakers contain a pointer to the TaskStorage, and wakers can live forever.

Ah, that makes sense.  Yes, that seems difficult to avoid without alloc.

I suppose one possible option would be to have a separate storage space for this (at which point you are basically implementing alloc, although it could be a pretty simple pool allocator since everything is fixed size).  In this separate storage space you could keep an entry for each task which is basically a (refcount, task pointer), and the task could reset its pointer to null when the task is dropped.  The downside is that you would need to specify the size of this storage pool at compile time.  If you run out of space in this table at runtime, select() could always fall back to the naive approach (although this adds complexity by having 2 code paths).  I suppose the added overhead and complexity of this approach might not be worthwhile, though.

Thanks for the feedback!

Dirbaio, 2024-04-09:
if you add a `println` to `poll()` you'll see every time `periodic_loop()` requests a wake, all 3 futures get polled:
https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=8f3f5eb1921d24ae67f10209c21b5b25