embassy-rs/embassy #2798: Configuring pull-up for SPI MISO pin on STM-32

natrow, 2024-04-09:
For some applications, a pull-up resistor is required on the MISO pin of an SPI bus. Is it possible to use one of the STM32 internal pull-ups for this application? It seems like this would be the relevant line of code:

https://github.com/embassy-rs/embassy/blob/c575c7dc6cab49a99f36eedbdd41e97868e1102b/embassy-stm32/src/spi/mod.rs#L119

If there was an API change to allow passing a `Pull` input to the spi config struct, that would be fantastic.

Dirbaio, 2024-04-26:
pull requests welcome. ideally it'd work like the pulls in i2c, where you put them in Config. It should be easy to add.

joelsa, 2024-05-15:
I am using the MCP33131-10-E/MS, which does not use the MOSI pin and needs a pull-up on MISO in order to monitor the status of the recalibration. Up until now, I hacked my way around it, but decided to take a shot at adding this in: https://github.com/embassy-rs/embassy/pull/2943

closed 2024-05-24 by Dirbaio