embassy-rs/embassy #2830: Allow using embassy-time Timer with embassy_futures::block_on

brt-adam-snaider, 2024-04-16:
I'm happy to send a PR to do this, but I'm just not sure if there's a reason this wouldn't work to begin with.

The reason this would be useful for us is that we write tests as async functions but we execute them serially using embassy_futures::block_on.

For the record, this is the panic message I get when I try to do this

```
ERROR panicked at 'Found waker not created by the Embassy executor. `embassy_time::Timer` only works with the Embassy executor.'
```

Dirbaio, 2024-04-17:
You can fix this by enabling feature `generic-queue` in `embassy-time` and disabling feature `integrated-timers` in `embassy-executor`.

brt-adam-snaider, 2024-04-17:
Works for me, thank you!

closed 2024-04-17 by brt-adam-snaider